nasa/cFS #2: core-linux sample build fails to start

william-marchetto, 2019-05-31:
I've tried running the cFE core-linux sample build in fresh VMs of CentOS 7, Ubuntu 16.04, and Ubuntu 18.04, and am getting the same error each time I try to run core-linux.bin:

```
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 66
CFE_PSP: Default CPU Name: CPU1
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2029-107-16:47:31.25462 POWER ON RESET due to Power Cycle (Power Cycle).
2029-107-16:47:31.25465 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2029-107-16:47:31.25466 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2029-107-16:47:31.25467 ES Startup: Starting Object Creation calls.
2029-107-16:47:31.25467 ES Startup: Calling CFE_ES_CDSEarlyInit
2029-107-16:47:31.25470 ES Startup: Calling CFE_EVS_EarlyInit
2029-107-16:47:31.25474 Event Log cleared following power-on reset
2029-107-16:47:31.25474 ES Startup: Calling CFE_SB_EarlyInit
2029-107-16:47:31.25477 ES Startup: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 ES Startup: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00008 ES Startup: Calling CFE_FS_EarlyInit
1980-012-14:03:20.00016 ES Startup: OS_TaskCreate error creating core App: CFE_EVS: EC = 0xFFFFFFFF
CFE_PSP_Panic Called with error code = 0x00000006. Exiting.
The cFE could not start.
```

I get a few warnings when running make, but they don't seem to be a problem:
```
>>> [POST-BUILD] COPY BUILT FILES TO <CFS_PROM_APPS> <<<
 >> Create Directories <<
mkdir -p ./exe/cf
mkdir -p ./exe/cf/apps
mkdir -p ./exe/cf/log
mkdir -p ./exe/cf/upload
mkdir -p ./exe/cf/download
mkdir -p ./exe/ram

 >> Change Permissions <<
chmod 775 ./exe/*.bin
chmod 644 ./exe/*.so
chmod 644 ./exe/*.tbl
chmod: cannot access './exe/*.tbl': No such file or directory
../cfs.mak:232: recipe for target 'cfs_copy_installs' failed
make: [cfs_copy_installs] Error 1 (ignored)
chmod 644 ./exe/*.scr

 >> Copy Files <<
'./exe/core-linux.bin' -> './exe/cf/apps/core-linux.bin'
'./exe/ci_lab.so' -> './exe/cf/apps/ci_lab.so'
'./exe/sample_app.so' -> './exe/cf/apps/sample_app.so'
'./exe/sample_lib.so' -> './exe/cf/apps/sample_lib.so'
'./exe/sch_lab.so' -> './exe/cf/apps/sch_lab.so'
'./exe/to_lab.so' -> './exe/cf/apps/to_lab.so'
../cfs.mak:232: recipe for target 'cfs_copy_installs' failed
make: [cfs_copy_installs] Error 1 (ignored)

../cfs.mak:232: recipe for target 'cfs_copy_installs' failed
make: [cfs_copy_installs] Error 1 (ignored)
'./exe/cfe_es_startup.scr' -> './exe/cf/apps/cfe_es_startup.scr'

>>> DONE! <<<
```
There are no *.tbl files, which is causing the first couple warnings, and ./exe/cf/apps/cfe_es_startup.scr exists, so that seems to be fine as well.

I've tried this with the 6.5.0a release and building from master and see the same boot error.

skliper, 2019-05-31:
Have you tried the rc-6.6.0a branch at https://github.com/nasa/cFS and associated instructions?  Should work on most standard linux workstations.

william-marchetto, 2019-06-20:
I was attempting to run cFE within docker. It looks like docker doesn't provide all of the OS-level control cFE requires. Runs fine on a normal virtual machine.

closed 2019-06-20 by william-marchetto