nasa/cFS #3: CMake reports that gcc is broken

shinn16, 2019-07-24:
There was an issue using cmake to perform any type of build operation (clean, prep, make, etc), it kept reporting that there was something wrong with gcc. This issue was that the following was missing from /cfe/CMakeLists.txt:

INCLUDE(CMakeForceCompiler)
CMAKE_FORCE_C_COMPILER(gcc GNU)
CMAKE_FORCE_CXX_COMPILER(g++ GNU)

Including these lines at the top of file fixed my issue. I am running CentOS 2.6.32-754.17.1.el6.x86_64

jphickey, 2019-07-24:
There should not be a need to force the compiler as a general case.  Can you comment further on the error details? There should be a log file that indicates why it thinks your compiler is broken and what the specific issue is.

shinn16, 2019-07-25:
After cloning a clean copy of the repo, I was unable to reproduce the issue. I do not have the log files from when the issue originally occurred as I deleted the repo and cloned a clean copy before attempting to force the compiler to ensure that I was working with a clean build directory. The error that was printed out to the terminal was very non-specific, it said "The C compiler "/usr/bin/gcc" is not able to compile a simple test program." This was probably just an issue with my local machine.

closed 2019-07-25 by shinn16