nasa/cFS #4: Examples for application testing

BaldBeacon, 2019-08-06:
Since the release of cFE 6.6, there hasn't been apps that give an example of the way tests should be defined in the CMakeLists file. Previously, when UTAssert was part of the tools directory rather than being part of the OSAL, this definition would exist inside of the fsw/for_build/Makefile but this doesn't cleanly translate to cFE 6.6's cmake build system.

The arch_build.cmake file located in the cmake directory seems to have a function [add_unit_test_exe](https://github.com/nasa/cFE/blob/99038d5446aec902e8cdbe14d762a86ea4e19e1c/cmake/arch_build.cmake#L202) but I couldn't get this to work when calling it from the simple app's CMakeList.txt. An explanation of what needs to be done in either the target.cmake or CMakeLists file for an app or an example of an app using UT for it's testing would be appreciated.

ejtimmon, 2019-08-06:
Yes, this is a problem.  I'm working on getting the GSFC cFS apps fully migrated to cFE 6.6/Cmake.  Hopefully I will have an example to release soon.

BaldBeacon, 2019-08-06:
Cool, thanks for the confirmation.

skliper, 2019-10-02:
CCB 20191001 - Plan to incorporate in sample_app and app developer's guide also

skliper, 2020-01-22:
See https://github.com/nasa/sample_app/pull/26 and https://github.com/nasa/sample_lib/pull/11 as a work in progress.

closed 2020-01-22 by skliper

BaldBeacon, 2020-01-22:
What was the duplicate issue #?

skliper, 2020-01-22:
https://github.com/nasa/sample_lib/issues/9 and https://github.com/nasa/sample_app/issues/8, both linked to the pull requests.