nasa/cFS #5: PSP location and CMakeLists

cmanderino, 2019-08-11:
Just upgrading some cFS mods we make in the lab with the software bus to this cFS build, but I've been running into some new pathing issues.

Notably, in case anyone else gets hung up on it, I have some CMake fail to finds:

**ln 41 in global_function.cmake reads:**

_get_filename_component(MISSION_SOURCE_DIR "${CMAKE_SOURCE_DIR}/cFS/cFE" ABSOLUTE)_

**From this repo "psp" belongs to "cFS" but, this seems problematic in arch_build.cmake ln 26**

_include_directories(${MISSION_SOURCE_DIR}/psp/fsw/shared)_

**Also: "cFE" v "cfe" as the submodule in names sometimes gets in the way**

skliper, 2019-08-12:
Could you add links to the code you are referring to?  I don't see it matching what's currently in the repo, where the only reference to get_filename_component in global_function.cmake in the cFE master branch is:

https://github.com/nasa/cFE/blob/99038d5446aec902e8cdbe14d762a86ea4e19e1c/cmake/global_functions.cmake#L65

It looks like there is a similar line to what you describe at:

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/cmake/mission_build.cmake#L38-L41

or

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/cmake/mission_build.cmake#L148-L152

but these all work fine for me.  What exactly is your issue?

Not clear what your issue with the include_directories line is?  What about this doesn't work?

Note I think your 3rd issue has already been solved, current .gitmodule lines:
https://github.com/nasa/cFS/blob/d42368f4b6a48df09d42a1a24a13609c8b325ffe/.gitmodules#L1-L4

cmanderino, 2019-08-26:
> Any updates on the issues?

Heya,

Sorry looking at 6.6.0 is on my todo list this week. I just had to jump
over to a few other projects before coming back around. I will let you know
how it goes!

Thanks,
Chris M.

skliper, 2020-07-23:
Unable to recreate issue and likely against old code base.  Feel free to reopen if issue can be recreated and version information identified.

closed 2020-07-23 by skliper