nasa/cFS #8: use appid with perfid for apps

CDKnightNASA, 2019-09-04:
Currently apps define their perfid when using the performance logging framework, but there's no mechanism currently to ensure that the perfid is unique to that application and could result in confused perf logs.

One option would be to use the appid generated when the application is registered (but will need to be reported somehow so that a performance analysis tool can crosswalk from runtime-generated perfid's to the applications.)

skliper, 2019-09-04:
@CDKnightNASA Could you use the feature request issue template?

CDKnightNASA, 2019-09-04:
> @CDKnightNASA Could you use the feature request issue template?

Now I know what you mean by template. ;D

skliper, 2019-09-04:
Any updates to IO would need to be coordinated with java performance monitor tool which I don't think has been open sourced yet.

Maybe a redesign instead of using defines, an app could register with ES to get however many performance IDs are needed.  Let ES manage it, and ES could output a file for input to the performance monitor tool with the index, app/service, and short text used by the app as part of the registration?

'''
CFE_ES_PerfLogRegister(&MyMainPerfID, "Main loop");

while ()
{
  CFE_ES_PerfLogExit(MyMainPerfID);
...
'''

mprather, 2020-04-06:
I think this is an interesting idea. 

One problem with the current approach is the lack of clear documentation on what/how this should be utilized. Going through the sample app and "cFE Application Developers Guide.doc", it's unclear if the value should be unique to the app, unique across the apps on the core, or globally unique.  The doc doesn't even mention anything about the ID and only shows the sample app code. The sample app simply sets and uses the value. From that perspective, there's very little to go on.

Then today, I realized there's a .md with the same name of as the .doc. Fortunately, it has some more insight and actually leads me to believe that that a "register" function should effectively account for namespacing the id. 

It seems that if an app is supposed to maintain ids that meet this statement -> "(applications) must define a unique performance id (or perfid) for each segment of code to monitor", then why would there ever be a conflict with another app?