nasa/cFS #11: Improve version control, reporting, and integrity support to enhance security

skliper, 2019-09-13:
**Is your feature request related to a problem? Please describe.**
Elements could be modified between original generation and use without easily being noticed.  A holistic design approach to version control, reporting, integrity should be considered to support the various use cases.

Currently state:
1. Repo information (git describe in a subset of repos) is gathered during build and reported on execution for top branch, cFE and OSAL- need to expand this to include all the elements (apps, libs, psp, etc)
2. Build information is gathered during build and reported on execution - user name and date for last full build covering cFE (doesn't cover uploaded apps or elements built separately/after the original full build)
3. Version information updated by hand and reported at execution time for most of cFS - elf2cfetbl and tblCRCtool don't report version information, hasn't been consistently updated
4. Checksums are typically just calculated and reported at load/execution, doesn't cover libs, really should employ digital signatures that can be verified

**Describe the solution you'd like**
See above suggestions

**Describe alternatives you've considered**
None, but should discuss with security experts and come up with a complete, well vetted solution.

**Additional context**
Likely very beneficial to community requiring a more robust security approach

**Requester Info**
Jacob Hageman/NASA-GSFC

skliper, 2019-09-13:
@jwilmot - heads up for discussion

skliper, 2019-09-16:
This topic also needs to cover the onboard integrity verification (currently CS) and related enhancements.

skliper, 2019-10-02:
CCB 20191001 - Brought up topic, will likely depend on funding

astrogeco, 2020-08-20:
@ArielSAdamsNASA this might interest you