nasa/cFS #17: Table Load Failures

JimKaidyNASA, 2019-10-10:
My app, named CNT_APP starts running and while there are tables in the app, it appears that CFE_TBL attempts to load those tables and fails, causing CNT_APP to error out.  Is there a missing step that I need to perform (load tables?) or is this indicative of a bug or misconfiguration?

[image]

jphickey, 2019-10-11:
Please use the mailing list for this type question.  This is almost certainly a mis-configuration and not a bug or issue in CFE_TBL.  

Note that error code 0xCC000020 is CFE_TBL_ERR_FILE_FOR_WRONG_TABLE (see cfe_error.h).  It means the name embedded in your table file does not match the table name you are trying to load.

Recommend closing this.  CFE_TBL is working correctly.

JimKaidyNASA, 2019-10-11:
Thank you for pointing out the error code.  I see the extensive error capture in this and will use in the future.  As to the reason for this one, I don't understand why the header file is different than the name of the table.

jphickey, 2019-10-11:
It is a simple strcmp, between the name of the table in your app's `CFE_TBL_Register()`  call and the table name specified in the table file definition.  These two strings just need to match.  Check for case mismatches, spaces, or similar-looking chars between your app code and the actual table definition file.

Closing this as not a bug.

closed 2019-10-11 by jphickey