nasa/cFS #20: I'm trying to get the syntax right for the validation command.  I've been running gdb to look at the contents of the message.  All I get is a cryptic invalid msg length error.  With the following from the cFS-GroundSystem/Guid, instructions, I know I am not interpreting it right.

JimKaidyNASA, 2019-10-24:
I'm trying to get the syntax right for the validation command.  I've been running gdb to look at the contents of the message.  All I get is a cryptic invalid msg length error.  With the following from the cFS-GroundSystem/Guid, instructions, I know I am not interpreting it right.

[image]

[image]

[image]

_Originally posted by @JimKaidyNASA in https://github.com/nasa/ECI/issues/42#issuecomment-545620809_

JimKaidyNASA, 2019-10-24:
I retrieved the latest GroundSystem-master.zip and did a new make.

I sent the command as before and still seeing the Invalid msg length error.

skliper, 2019-10-24:
No surprise since there's likely cross versioning going on here.  You'll need to make the command definition match the software.

JimKaidyNASA, 2019-10-24:
I'm using what is written in the Description box for CFE_TBL_INACTIVE and ACTIVE.  Where can I find the proper syntax written for the command?

JimKaidyNASA, 2019-10-24:
It appears in the debugger that the CFE_TBLE_ValidateCmd is not actually being called.  I've set a breakpoint dwelling on this call from the cfe_tbl_task.c and the function is never entered although the CommandCode is set properly when the error occurs.  Since the CommandCode 4 is being set properly it makes sense that the Invalid msg length error is being thrown in the CFE_EVS_SendEvent since the CFE_TBL_ValidateCmd is actually never executed.

skliper, 2019-10-25:
It's not a problem with the syntax you are using.  It's a mismatch between the definition of the packet in the ground system and the definition of the packet in the cFE.  You need to make:

https://github.com/nasa/cFS-GroundSystem/blob/master/Subsystems/cmdGui/ParameterFiles/struct_c_f_e___t_b_l___validate_cmd__t

match whatever it is your version of table services has defined.  Likely in a file similar to what's in the latest version:

https://github.com/nasa/cFE/blob/777f20a12c7592f2048621a9b19eeda919cf8b41/fsw/cfe-core/src/inc/cfe_tbl_msg.h#L553-L569

JimKaidyNASA, 2019-10-25:
Ok.  I'll be the first to admit I am unclear about how the packet definition is translated back into code.  I see uint16 and char which are consistent with the code.  I also see comments consistent with the commented code.  Not sure if lp0 and lp1 are lines, p2, p3, etc are spacing, S versus aS some time of string designation,  aa is some type of grouping of features.  All this points to the command window as far as I can tell, which is much more information than is in the code.

skliper, 2019-10-25:
The most likely issue is the string size doesn't match your code.  It's the aS'38' line.  Make the number match the size of TableName in CFE_TBL_ValidateCmd_Payload_t.  Or just modify it such that you don't get the error (if reported length is 10, and 20 is expected, add 10 to the number).

JimKaidyNASA, 2019-10-25:
Ok that worked.  I modified S38 to S40 since the length discrepancy was off by 2.  I found that I had to do the same thing for the Table Activate command where I modified the value from S38 to S40 in struct_c_f_e_t_b_l__activate_cmd__t.

Is it correct to say that any user who is using cFE 6.5 will have to make these two mods to the GroundSystem parameters to have these (and possibly others) work properly?

JimKaidyNASA, 2019-10-25:
Correction aS38 to aS40

skliper, 2019-10-28:
> Is it correct to say that any user who is using cFE 6.5 will have to make these two mods to the GroundSystem parameters to have these (and possibly others) work properly?

Yes.  In general it's on the user to handle any cross versioning or customization of sizes issues.  The open source community typically focused on getting the system to work with all the current versions for the default configuration.

closed 2019-10-28 by skliper