nasa/cFS #21: Syntax for Table Dump Command

JimKaidyNASA, 2019-10-28:
When attempting to dump a table the format for the python GroundSystem is unclear.

[image]

My attempt syntax is as follows.  It appears to concatenate the table and the table file name, making it hard to discern any proper convention for the arguments of the command.

[image]

astrogeco, 2020-03-02:
@JimKaidyNASA not sure if you've figured this out but for future questions use the mailing list 
https://lists.nasa.gov/mailman/listinfo/cfs-community

avan989, 2020-03-13:
@JimKaidyNASA This is a little late but check your parameter file for the dump command. Your length definition for the tableName is off, that is probably why it is getting concatenate like that. I recommend just redoing the entire "table" command for the Ground System, your length for the DumpFileName is probably off too. 

Reference Guide-GroundSystem.txt for instructions on how to generate command. Hope this help.

closed 2020-03-18 by astrogeco