nasa/cFS #24: OS_API_Init failure

SweeWarman, 2019-11-13:
I cloned this cFS repository with all its submodules, followed the instructions in the README for compilation. I get the following error when running the executable (core-cpu1). 

CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: OS_API_Init() failure

I am working on a linux workstation running ubuntu 18.04. Any insight would be helpful.

skliper, 2019-11-13:
Did you configure for permissive mode, as specified in the readme?  If not you need to run the code with elevated privileges.

Version information (did you pull master or a tag/release?), and your exact steps to recreate the issue would likely get you a more applicable response (information from the bug issue template is useful).

skliper, 2019-11-13:
Also may help to enable debug messages by defining OS_DEBUG_PRINTF in your default_osconfig.h.

SweeWarman, 2019-11-13:
Including the define for the permissive mode seems to have fixed this issue. Many thanks for the swift response.

closed 2019-11-13 by SweeWarman