nasa/cFS #25: CFE_ES_GetTaskInfo: Task ID Not Valid: 0

SweeWarman, 2019-11-14:
I get the following message when I try to publish a message on the software bus from within a call back function that is called by a timer. 

CFE_ES_GetTaskInfo: Task ID Not Valid: 0

Steps to reproduce the behavior:
[sample_app.txt](https://github.com/nasa/cFS/files/3848156/sample_app.txt)
- Download the attached sample_app.txt
- Rename .txt to .c
- Replace apps/sample_app/fsw/src/sample_app.c with the renamed sample_app.c file

compile and run core-cpu1.

**System observed on:**
 - x86_63
 - OS: [e.g. Linux 4.15]
 - Versions [cFS repository at commit hash: 044476b (current master branch as on 11/14)]

**Reporter Info**
NASA Langley/NIA

Any thoughts on how to resolve this is greatly appreciated.

jphickey, 2019-11-14:
This is correct/expected behavior.  A timer callback is a different task context.  Therefore when the `CFE_SB_SendMsg` call determines the calling context, it isn't from a thread/application that CFE ES knows about, and therefore you get this warning.

There are two alternative approaches:
- Create a semaphore and a child task.  Timer callback posts the semaphore.  Child task pends on the semaphore, then does any remaining work and sends the message.
- Use SCH to schedule your wakeup events rather than creating an OSAL timer directly.  This is basically the same idea as above but uses the message pipe instead of a semaphore to perform wakeup.

closed 2019-11-14 by SweeWarman

jphickey, 2019-11-14:
This may be worth a quick discussion - maybe we should consider implementing a more descriptive error/event message than "Task ID Not Valid: 0"?  This error string is not very helpful or informative and probably could be better.

skliper, 2019-11-15:
I definitely support a more descriptive response.

mlouielu, 2020-06-09:
Is it necessary for SB to know which calling context it is inside? (e.g. is `GetTaskInfo()`, `CFE_ES_GetTaskInfo()`  a must?)