nasa/cFS #28: Automate a deployment of docs to GitHub pages

BaldBeacon, 2019-12-09:
Having to build and open the documentation locally adds an extra step that could be mitigated by pushing the documentation of the current master commit to GitHub pages.

The process of uploading to GitHub pages is documented [here](https://pages.github.com/)

skliper, 2020-03-16:
Related to https://github.com/nasa/cFE/issues/560

From Circle CI instructions at https://circleci.com/blog/deploying-documentation-to-github-pages-with-continuous-integration/.  Travis CI is likely similar.

Certainly a useful way to keep our doxygen documentation up to date.  Need to trade approaches.

skliper, 2020-03-16:
Travis guide: https://docs.travis-ci.com/user/deployment/pages/

PR #61 (2020-04-01): Update CI for deploy docs, warnings, and simplify
**Describe the contribution**
Fix #28, Deploy documentation
Fix #60, config warnings except for the new ones
introduced by using the latest deployment version
Fix #47, removes permissive mode hack and updates README.md
Adds enforce of no doxygen warnings for osal and users guide

**Testing performed**
Steps taken to test the contribution:
1. CI test

**Expected behavior changes**
Old CI script warnings resolved (2 remain based on using development version)
Now enforces no doxygen warnings for user guides
Will deploy documentation to gh_pages br [... cut]
commit 928e3c0e (2020-04-01): Update CI for deploy docs, warnings, and simplify

Fix #28, Deploy documentation
Fix #60, config warnings except for the new ones
introduced by using the latest deployment version
Fix #47, removes permissive mode hack and updates README.md
Adds enforce of no doxygen warnings for osal and users guide
PR #62 (2020-04-03): Integration Candidate 2020-03-25
**Describe the contribution**
Integration Candidate 2020-03-25,
incorporates changes to sample_app, cFE, and OSAL

Fix #28, #60, and #47, 

**Testing performed**
PR #61 
Bundle CI

**Expected behavior changes**
See related Integration Candidate PRs
nasa/osal#52
nasa/osal#401
nasa/cFE#579

Updates to Bundle CI:
- Deploy documentation, 
- config warnings
- removes permissive mode hack
- Adds enforce of no doxygen warnings for osal and users guide

**System(s) tested on**
Bundle CI - Ubuntu:Bionic

**Code contibutions**
Jacob Hageman - NASA/GSFC
Gerardo E. Cruz-Or [... cut]
closed 2020-04-03 by astrogeco