nasa/cFS #46: WSL Support

Flern, 2020-02-19:
Hello:

I cloned and built using the README instructions (both permissive and non-permissive, and launched with / without sudo) and I get the OS_API_INIT error on startup.

I'm running the Ubuntu distro on Windows Subsystem for Linux.  Surely that's the root cause.  No urgent need to get it running, just wondered if this deployment is something on the roadmap for future support.

If more specific troubleshooting is desired, let me know how to gather the information and I'll get it for you.

stanislaw, 2020-02-19:
I am not a CFS maintainer so I might be wrong and also I am not a Windows user but I have set up cFS on Ubuntu recently. I had a few initialization-related errors so maybe the errors that you are seeing are not Windows- but Linux-specific.

First of all, you can enable the debug lines:

```
/*
 * If OS_DEBUG_PRINTF is defined, this will enable the "OS_DEBUG" statements in the code
 * This should be left disabled in a normal build as it may affect real time performance as
 * well as producing extra console output.
 */
 
// #undef OS_DEBUG_PRINTF
#define OS_DEBUG_PRINTF
```

After doing that and following the CFS recommendations you will most likely have to do:

1) Setting up necessary RT-related permissions

```
# https://stackoverflow.com/a/10705753/598057
your_user        hard    rtprio          99
your_user        soft    rtprio          99
```

2) Increasing the limit for POSIX Message Queues

```
sudo su - # just sudo does not seem to work
echo 100 > /proc/sys/fs/mqueue/msg_max
```

I think this was it for me to run it on Ubuntu. Hope this helps.

Flern, 2020-02-20:
Thank you for the tips!

I followed your Ubuntu setup steps.  The only difference between your environment and my WSL is that there is no `mqueue` in `/proc/sys/fs`

After following the steps, I still fail to initialize:
```
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
CFE_PSP: Starting the cFE with a POWER ON reset.
Error: pthread_mutex_init failed: Operation not supported
OS_API_Impl_Init(0x1) failed to initialize: -1
CFE_PSP: OS_API_Init() failure
```
Looks like this is coming from line 1498 of my copy of osapi.c.  

Further investigation shows that I'm running WSL version 1 with Windows 10 build:
`Microsoft Windows [Version 10.0.17134.1304]`

This indicates that I likely have the same issue described here:  [https://github.com/microsoft/WSL/issues/3449](https://github.com/microsoft/WSL/issues/3449)

Thank you again for your help.  Looks like my WSL isn't going to work with this until my Win10 gets upgraded.

closed 2020-02-20 by Flern

skliper, 2020-02-20:
@Flern thanks for the update, definitely keep us in the loop if you get this working.  I've been interested in WSL myself.

jphickey, 2020-02-24:
Correct - I have tested on WSL in the past and can confirm that that WSL lacks POSIX mqueue support which is what prevents it from running in this environment.  This would be resolved by implementing a semaphore-based mqueue alternative within OSAL.  This would also be helpful for OSX and there may even be an existing enhancement ticket for this.

skliper, 2020-03-02:
https://github.com/nasa/osal/issues/73