nasa/cFS #50: cFE API documentation ...

bgmarchand, 2020-02-26:
There are several places across various websites and documents that allude to the cFE API documentation, but I can't seem to find it anywhere. Does that actually exist? If so, could you point me to where that is? The application developer's guide is in the Git repository, but I'm specifically talking about the API documentation for cFE.

skliper, 2020-03-02:
The API is part of the user's guide that gets generated using doxygen.  Typically we'd attach the generated file for official releases but it has been a while since we've had one.  https://github.com/nasa/cFS/pull/40/files contains steps to build the documentation, but we've currently got a bug in creating the pdf (although html does work).  I'll work to get the bug resolved and update once complete.

skliper, 2020-03-03:
[refman.pdf](https://github.com/nasa/cFS/files/4282817/refman.pdf)

See attached for the current autogenerated documentation that includes the cFE APIs.  It is undergoing updates, but at least a good start.

closed 2020-03-03 by skliper

bgmarchand, 2020-03-03:
Wonderful. Thank you! This is exactly what I was looking for.

skliper, 2020-03-03:
Note there are a few out of date comments, but I'm working them now.