nasa/cFS #63: Reference autogenerated docs

skliper, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
See #61 
No current references to the generated documents

**Describe the solution you'd like**
Once the docs are deployed the first time (after push of #61 to master and CI completes) add references from at least README.md.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-10:
Just add references in README.md to the documents in the gh_pages branch.

JandlynBentley-at-NASA, 2020-06-09:
I'm looking for clarification (I'm new to cFS).  Which references are you specifically referring to?

skliper, 2020-06-11:
https://github.com/nasa/cFS/tree/gh-pages

JandlynBentley-at-NASA, 2020-06-12:
Apologies for not being more specific, I meant which references in the README should be added to the cFE and OSAL user guides in the gh pages branch?

skliper, 2020-06-12:
I may not be understanding the question, but the request is to add references in the README.md to the user's guides.  Just add a "References to auto-generated documentation" section, and links to the two files.

PR #99 (2020-06-23): Fix #63, Added the references (links) to the cFE and OSAL user guides from the…
**Describe the contribution**
Added the cFE and OSAL user guides' references (links) from the gh-pages branch to the cFS bundle README.md. 

Resolves: #63 

**Contributor Info - All information REQUIRED for consideration of pull request**
Contributed by Jandlyn Bentley, NASA GSFC
commit e74a4855 (2020-06-23): Fix #63, Added the references (links) to the cFE and OSAL user guides from the… (#99)

* Added the references (links) to the cFE and OSAL user guides from the gh-pages branch.

* Moved the references to the top of the document before release notes.
PR #106 (2020-06-25): Integration Candidate: 2020-06-17
**Describe the contribution**
Combines
https://github.com/nasa/cFE/pull/756
https://github.com/nasa/osal/pull/521
https://github.com/nasa/cFS-GroundSystem/pull/107

Also, 
Fix #63

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/106/checks

**Expected behavior changes**
PR #99 - Added the references (links) to the cFE and OSAL user guides from the gh-pages branch.
PR #103 - Update PyQt reference in docs

See submodules PRs for more details

**System(s) tested on**
Ubuntu:Bionic

**Additional context**
None

**Code contibutions**
N/A

**Contr [... cut]
closed 2020-06-25 by astrogeco