nasa/cFS #65: Add build verification scripts (CTF based)

skliper, 2020-04-10:
**Is your feature request related to a problem? Please describe.**
No build verification scripts for command requirements

**Describe the solution you'd like**
Add build verification scripts

**Describe alternatives you've considered**
None.

**Additional context**
Need to discuss how we approach this.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-10:
Do we add them to cFS, a separate new repo, cFE repo?  It's part of providing a certifiable package and this is an opportunity for open source to leverage the Gateway work.

skliper, 2020-04-15:
CCB 20200415 - Recommendation to add to cFE, these are specific to cFE

astrogeco, 2020-04-20:
@skliper should I "move" this issue to the cfe repo?