nasa/cFS #76: move cppcheck from .travis.yml to Makefile.sample

CDKnightNASA, 2020-04-30:
It would be convenient to be able to run cppcheck in our checked out repo's without having to fish it out of the .travis.yml--what about moving it from .travis.yml to Makefile.sample and then having .travis.yml just call "make cppcheck"?

For that matter might be useful to add Make rules to "apt install" the required packages.

**Requester Info**
Christopher.D.Knight@nasa.gov

skliper, 2020-04-30:
The make cppcheck could also do all the more stringent sub-repo checks.

CDKnightNASA, 2020-04-30:
> The make cppcheck could also do all the more stringent sub-repo checks.

Kinda been thinking of a number of "semantic", workflow-oriented make commands...such as:
* `make setup-env`
  * would apt install the required packages
  * would configure environment variables
* `make ticket-branch`
  * would ensure it's on a forked repo
  * would create the branch with the name fix-<N>-<desc>
* `make commit-for-pull` that would execute a number of subtasks like:
  * `make cppcheck` which would store the cppcheck output in the branch and would alert on errors
  * `make test` which would store the coverage test output in the branch and would alert on a decrease in the coverage %
  * check that there's no other commits on this branch (or suggest a squash)
  * check that the branch name follows our format of 'fix-<N>-<desc>`
  * check if the fork/branch is "current" against nasa/<repo>/master.
  * warn if there are no unit test, stub, doc changes in the commit
  * commit with the "fix #<N> - comment" formatted message.
Anyways, food for thought.

jphickey, 2020-05-01:
I like the idea of a make target to run cppcheck.

astrogeco, 2020-05-05:
Check out my very simple implementation in #81 not quite the same but I think it's in a similar direction

CDKnightNASA, 2020-05-05:
> Check out my very simple implementation in #81 not quite the same but I think it's in a similar direction

Works for me, you can tie this ticket to that pull req so you can consider this ticket closed when the pull req is done.

skliper, 2020-05-05:
I'd prefer if the cppcheck was added to the makefile, and covered all the stricter submodule cppchecks...