nasa/cFS #90: Create directory structure for pushing CTF scripts

dmknutsen, 2020-05-26:
**Is your feature request related to a problem? Please describe.**
Need to create a location to push CTF scripts and plugins to.

**Requester Info**
Dan Knutsen
NASA Goddard

astrogeco, 2022-05-03:
Reviving this since we probably want to incorporate cTF tests into Draco

@tngo67 you all are using a directory named `ctf_tests` in apps right? What do you think of the structure suggested above?

tngo67, 2022-05-03:
@astrogeco  I like it, but with a minor name change: from "json" to "ccdd" since both ctf & ccdd files are in json format

Our current directory is <app>/fsw/ctf_test, but we can change that to be more consistent across all common apps.

skliper, 2022-05-04:
I wonder if there are more generic names for these, since linking directly to the tool may be over-constrictive.  Really it's the database dfn and requirements verification tests, no?  If the json comes from CCDD or any other tool shouldn't really matter, similar to the json scripts that implement the tests it doesn't need to only be ctf but could be any tool that can ingest the json files.  I also wouldn't have the database under verification tests... they both make more sense to me at the top level.

skliper, 2022-05-04:
That said, we've also got the eds descriptions in cfe/modules/core_api/eds... maybe instead of having multiple "database" files, we just provide the recipe for conversion?  The problem I see is when we have multiple "truths" they invariably end up conflicting.  Better to pick one truth and generate the other products from that truth.

tngo67, 2022-05-04:
I'm thinking, "ctf" because the tests are implemented in ctf-specific json schema.  They won't work with other testing tool/framework as-is.  "ccdd" for Command & Data Dictionary (not the ccdd tool, we could remove a 'c' in the name) also in the ctf-specific json schema.  We do plan to develop CTF tools/scripts to do the conversion from the GW xtce, so we can plan to do the same for cFS EDS.

I would also add a plug-ins sub-dir.