nasa/cFS #93: Small hiccup when installing the cFS bundle

JandlynBentley-at-NASA, 2020-05-28:
In following the steps laid out in the cFS bundle's readme, everything up through the "Build and Run" section worked fine. However, there was a small hiccup that occurred in the "Send commands, receive telemetry" section after installing PyQt5 (instead of the prescribed PyQt4) and PyZMQ.  

I changed directory back to the main cFS location, and tried to follow the next insruction: cd tools/cFS-GroundSystem/Subsystems/cmdUtil, but I got this error message: No such file or directory.  I tried installing CmdUtils via pip3, then re-tried the cd tools/cFS-GroundSystem/Subsystems/cmdUtil command again and still the same error occurred.  Skipping to the next line: python3 GroundSystem.py was unsurprisingly met with: pythons3: can’t open file ‘GroundSystem.py’: [Errno 2] No such file or directory.  I tried closing out the terminal and opening a new one, and retried cd tools/cFS-GroundSystem/Subsystems/cmdUtil, but still no luck.  Later after trying a new terminal again, I thought maybe PyQt5 instead of PyQt4 was a problem (but nothing changed).

And finally, trying a new terminal window one last time, I was able to get cd tools/cFS-GroundSystem/Subsystems/cmdUtil and all of the subsequent commands to work, and launch the ground system gui.  I am not sure why, after changing nothing from the last attempt, it would work this time.

This was attempted on a Linux VM with Ubuntu 20.04.

skliper, 2020-05-28:
Hard to tell what the issue was, maybe you weren't in the expected directory when executing the change directory command?  Let us know if you can recreate the issue, the steps to recreate, and then we may be able to update the instructions.

JandlynBentley-at-NASA, 2020-06-01:
**Update:**
I re-cloned the cFS bundle from this repository, still using Ubuntu 20.04 on the Linux VM.  I installed PyQt5 and PyZMQ before using the setup commands, as well as all other necessary software including python3, pip3, cmake, and git (gcc already pre-installed). After following through "Setup" and "Build and Run", again with no issues, I closed the terminal (because it no longer took commands at this point; ls, clear, and even exit did nothing) and opened a new one.  

After cd-ing into cFS, the cd tools/cFS-GroundSystem/Subsystems/cmdUtil, make, and cd ../.. instructions ran fine. However, when I tried to run python 3 GroundSystem.py, I got an interesting error:

"qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because the Qt platform plugin could not be initialized.  Reinstalling the application may fix this problem.

Available platform plugins are" eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)"

Unsuccessful attempts to fix the issue:
1) sudo apt-get libxkbcommon-x11-0
2) export QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb

**Successful attempt:**
Uninstalling pyqt5 (it was installed via pip3) followed by reinstalling pyqt5 by: 
sudo apt-get nstall python3-pyqt5

**Conclusion:** PyQt5 is finicky; use sudo apt-get for installation.

JandlynBentley-at-NASA, 2020-06-26:
It would seem that the following will resolve the issues:

1. Upon a fresh clone, make sure make, cmake, gcc, and git are installed.
2. Once the initial setup (build and run) is complete, open a new terminal.
3. There, the items for the ground system are very picky in how they want to be installed. It must be done this way:
   - $ sudo apt-get install python3-pyqt5
   - $ sudo apt-get install python3-zmq
   - $ sudo apt-get install libcanberra-gtk-module

I've written up a detailed improvements recommendation report and will submit it to Gerardo.

closed 2020-06-26 by JandlynBentley-at-NASA

skliper, 2020-06-29:
Note I typically create a virtual environment (venv) for the ground system python installs, such that they don't interfere with system or other python use.  Also, either Ctrl-C or sending a reset command from cmdUtil will stop cFS from running.

skliper, 2020-06-29:
Note make, cmake, gcc, and git (and many other useful tools) often can be installed via a development tools group or similar depending on the distribution.