nasa/cFS #95: CI - build multiple targets

CDKnightNASA, 2020-06-02:
**Is your feature request related to a problem? Please describe.**
The build process becomes increasingly more complex when building for multiple targets. CI should build for at least two targets to ensure that the build system is functioning properly.

**Describe the solution you'd like**
Have CI create relevant build configuration to build for multiple targets.

**Requester Info**
Christopher.D.Knight@nasa.gov

skliper, 2020-07-27:
In light of the conversation in nasa/cFE#758, this needs more discussion as to how it should be implemented.

Some options
 - Make it a built in option as part of sample_defs, CI just sets a flag
 - poke in changes, or patch
 - add separate *_defs for CI with "special" configurations

CDKnightNASA, 2020-07-27:
> In light of the conversation in [nasa/cFE#758](https://github.com/nasa/cFE/pull/758), this needs more discussion as to how it should be implemented.
>
> Some options
>
> * Make it a built in option as [...]

Not sure if this is one of the options you describe above, but my original thought was that .travis.yml would have a simple sed script that would uncomment the CPU2 entries in targets.cmake prior to build/test/etc. Probably the most minimal impact.

jphickey, 2020-07-27:
Note the changes in nasa/cfe#776 to use name-based (rather than number-based) target definitions make this substantially easier to implement.  Shouldn't be a need for sed scripts to hack the targets.cmake file if we go this route for future CFE versions.

astrogeco, 2020-09-02:
[CCB 2020-09-02](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-September-02-2020) Keep open, see implementation in SBN for inspiration.

closed 2021-04-02 by astrogeco