nasa/cFS #104: Either FM unit testing is incorrect or an OSAL enumeration is incorrect

mbenson1, 2020-06-19:
**Describe the bug**
I don't know if this an FM bug or an OSAL bug.  I don't know what the intent on either side was.  FM is failing unit tests.  The first error indicates before OSAL 5.0.3-bv, FM determined if a directory entry was a directory by using the S_IFDIR macro directly, but had an #ifdef that allowed it to use an OSAL defined macro instead.  

```
#ifdef OS_FILESTAT_ISDIR
            if (OS_FILESTAT_ISDIR(FileStatus))
#else
            if (S_ISDIR(FileStatus.FileModeBits))
#endif
```

With OSAL 5.0.3-bv, the OS_FILESTAT_ISDIR macro is defined and FM is using the OS_FILESTAT_ISDIR macro.  The unit test sets the FileStatus.FileModeBits to 0040000 (0x4000), but the OS_FILESTAT_ISDIR tests equality against OS_FILESTAT_MODE_DIR (0x10000) so the unit tests fail.  I don't know if the OSAL developer intended to use the value expected by FM (0x4000), or if the FM should be:

from:
`filestats->FileModeBits = S_IFDIR;`

to:
```
#ifdef OS_FILESTAT_ISDIR
    filestats->FileModeBits = OS_FILESTAT_MODE_DIR;
#else
    filestats->FileModeBits = S_IFDIR;
#endif
```

Making the change above fixes these errors, but so does changing the enumeration from:
OS_FILESTAT_MODE_DIR =   0x10000
to
OS_FILESTAT_MODE_DIR =   0x4000

**To Reproduce**
Steps to reproduce the behavior:
1. Build FM unit tests with the ut_assert, hooks, and stubs from cFE 6.5.0a
2. Run FM unit tests

**Expected behavior**
FM unit tests should pass.

**System observed on:**
 - VirtualBox
 - Ubuntu 16.04
 - FM 2.5.2, OSAL 5.0.3-bv, app ut_assert, hooks, and stubs from cFE 6.5.0a

**Reporter Info**
Mathew Benson
Windhover Labs, LLC
mbenson@windhoverlabs.com

skliper, 2020-06-29:
@ejtimmon Is this resolved with the latest FM release?

astrogeco, 2022-05-03:
Issue moved to [nasa/FM #21](https://github.com/nasa/FM/issues/21) via [**ZenHub**](https://www.zenhub.com/)