nasa/cFS #105: Publish html version of users guide and leverage github pages

astrogeco, 2020-06-22:
**Is your feature request related to a problem? Please describe.**
The current deploy of the documentation is a pdf that while useful, could be better implemented as a webpage which doxygen already generates.

**Describe the solution you'd like**
Leverage github pages to host a "live" version of the users guide.

**Describe alternatives you've considered**
figure out how to get doxygen to generate markdown instead of html and use Jekyll or another static site generator

**Requester Info**
Gerardo E. Cruz-Ortiz, NASA-GSFC

skliper, 2020-06-22:
Just a note that my experience in using the HTML was dreadful.  You can't search, it's 100's (or 1000's) of random files... I hated that version and would prefer to not even make it to begin with.  The PDF in my experience is WAY more useful... with an index, etc.

astrogeco, 2020-06-22:
> Just a note that my experience in using the HTML was dreadful. You can't search, it's 100's (or 1000's) of random files... I hated that version and would prefer to not even make it to begin with. The PDF in my experience is WAY more useful... with an index, etc.

Maybe have both? That's sad that the html is bad. I was hoping it would be like https://pmarsceill.github.io/just-the-docs/

skliper, 2020-06-22:
Feel free if you find it useful.  I didn't, it's terrible.  The PDF looks way better.

skliper, 2020-07-23:
Can we close this, with the PDF's being sufficient?

astrogeco, 2020-07-23:
I don't think they are. What's the harm in leaving it open?

skliper, 2020-07-23:
No harm, just wanted to make sure it was still an active enhancement request.