nasa/cFS #107: CI - add test build/run with elevated privileges

skliper, 2020-06-29:
**Is your feature request related to a problem? Please describe.**
CI only builds/runs with user privileges

**Describe the solution you'd like**
Add an elevated privileges build/run

**Describe alternatives you've considered**
None.

**Additional context**
See #95 for additional update request

**Requester Info**
Jacob Hageman - NASA/GSFC

astrogeco, 2020-07-01:
Can you add more context to this? It seems like nasa/osal#528 needs it but I think we can add a more specific rationale in this issue
.

skliper, 2020-07-01:
Different logic/code path is exercised for a privileged user run.  Without including it in CI, we are not covering this code.  Basically low hanging fruit (easy change) to get more coverage from CI.  There is plenty of other code we don't cover (we don't do any build verification), but just a privileged run will at least avoid breakage like nasa/osal#528.

jphickey, 2020-07-01:
@astrogeco - this is why the original testing of PR nasa/osal#508 didn't show an issue - either locally for me or in CI.  It was noticed by a user who cloned the code and ran it as root.  When running as root some additional thread attributes are set that only a privileged user can do.

I'm working on some docker-based CI scripts in the background and this should be covered by those tests.  For the current CI it just means adding/changing at least one `make test` command to run in `sudo`, which should cover it.