nasa/cFS #110: OS_FPUExcSetMask and OS_FPUExcGetMask VxWorks OSAL unit tests fail

mbenson1, 2020-07-02:
**Describe the bug**
The VxWorks OSAL unit tests for OS_FPUExcSetMask and OS_FPUExcGetMask are CPU specific, not just OS specific.  These functions have conditionally compiled code that is conditional on whether or not the platform is the PPC440.  However, the unit test code does not.  So if the CPU is not a PPC440, these two unit tests will fail.  

**Expected behavior**
The unit test should pass, though I'm unsure whether the break is in the OSAL or the unit test code.

**System observed on:**
 - AiTech SP0-S
 - OS: VxWorks 6.9
 - OSAL 5.0.3-bv

**Reporter Info**
Mathew Benson
Windhover Labs

jphickey, 2020-07-02:
FYI - these OSAL functions are deprecated in the next release for this very reason.  Similar situation with interrupts.

mbenson1, 2020-07-02:
But I am on a tagged release, 5.0.3-bv.  Is it acceptable to have release versions fail unit tests?

jphickey, 2020-07-02:
> But I am on a tagged release, 5.0.3-bv. Is it acceptable to have release versions fail unit tests?

There's a few things here:

- `5.0.3-bv` is __not__ an official release tag.  The github descriptions of these have been (hopefully) updated to make that clearer that these were only testing/development tags and not releases.  You are welcome to test this, but it isn't an official release.
- The "AiTech SP0-S" isn't part of the GSFC testing matrix, only MCP750 is.  While we __do__ certainly want the code to be generic, some board specific stuff did creep its way into various places.  So yes, it IS (was?) a bug....
- the `5.0.3-bv` tag is almost a year old and there have been LOTS of bugfixes since then (including some major ones).  This is one such item that has already been address, which was the point of my previous comment.  The platform dependencies were noted in nasa/osal#366 and it was determined that these APIs don't really belong in OSAL (as you also noted) and the decision was made to deprecated them.

Again, the last 9-10 months have been heavily focused on cleaning up (both code and requirements), bug fixes, portability, and all those types of issues.  While we appreciate any and all bug reports, stuff from older tags is certainly likely to have been found already.

mbenson1, 2020-07-03:
Thanks.  I was using the osal version that the osal submodule in the cFS submodule pointed to on tag v6.7.3-bv.  When I started the upgrade, the newest non-"bv" tag was 6.6.0a, but now I see that v6.7.0 was just tagged earlier this week.  I'm switching to the v6.7.0 tagged release now.