nasa/cFS #112: pthread_attr_setstacksize error when running initial build

mentoc3000, 2020-07-15:
I have been playing around with cFS, having gotten it working by cloning this repository. After a couple of days of it working, I started getting the error below when running the software. I tried to start over with a new clone of this repo, but the same error message arises. I presume that it has something to do with my system settings, but I don't know what. Any guidance would be much appreciated!

```
$ ./core-cpu1 
OS_BSP_Initialize():Maximum user msg queue depth = 10
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
OS_Posix_GetSchedulerParams():188:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():188:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():374:Selected policy 2 for RT tasks, root task = 99
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
Aborted (core dumped)
```

<details>
  <summary>The full details will all commands and outputs can be found here.</summary>

```
XXX@XXX~/cFS$ git clone https://github.com/nasa/cFS.git test
Cloning into 'test'...
remote: Enumerating objects: 44, done.
remote: Counting objects: 100% (44/44), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 472 (delta 17), reused 31 (delta 9), pack-reused 428
Receiving objects: 100% (472/472), 3.87 MiB | 9.83 MiB/s, done.
Resolving deltas: 100% (217/217), done.
XXX@XXX~/cFS$ cd te
bash: cd: te: No such file or directory
XXX@XXX~/cFS$ cd test/
XXX@XXX~/cFS/test$ git submodule init
Submodule 'apps/ci_lab' (https://github.com/nasa/ci_lab.git) registered for path 'apps/ci_lab'
Submodule 'apps/sample_app' (https://github.com/nasa/sample_app.git) registered for path 'apps/sample_app'
Submodule 'apps/sample_lib' (https://github.com/nasa/sample_lib.git) registered for path 'apps/sample_lib'
Submodule 'apps/sch_lab' (https://github.com/nasa/sch_lab.git) registered for path 'apps/sch_lab'
Submodule 'apps/to_lab' (https://github.com/nasa/to_lab.git) registered for path 'apps/to_lab'
Submodule 'cfe' (https://github.com/nasa/cFE.git) registered for path 'cfe'
Submodule 'osal' (https://github.com/nasa/osal.git) registered for path 'osal'
Submodule 'psp' (https://github.com/nasa/PSP.git) registered for path 'psp'
Submodule 'tools/cFS-GroundSystem' (https://github.com/nasa/cFS-GroundSystem.git) registered for path 'tools/cFS-GroundSystem'
Submodule 'tools/elf2cfetbl' (https://github.com/nasa/elf2cfetbl.git) registered for path 'tools/elf2cfetbl'
Submodule 'tools/tblCRCTool' (https://github.com/nasa/tblCRCTool.git) registered for path 'tools/tblCRCTool'
XXX@XXX~/cFS/test$ git submodule update
Cloning into '/home/jp/cFS/test/apps/ci_lab'...
Cloning into '/home/jp/cFS/test/apps/sample_app'...
Cloning into '/home/jp/cFS/test/apps/sample_lib'...
Cloning into '/home/jp/cFS/test/apps/sch_lab'...
Cloning into '/home/jp/cFS/test/apps/to_lab'...
Cloning into '/home/jp/cFS/test/cfe'...
Cloning into '/home/jp/cFS/test/osal'...
Cloning into '/home/jp/cFS/test/psp'...
Cloning into '/home/jp/cFS/test/tools/cFS-GroundSystem'...
Cloning into '/home/jp/cFS/test/tools/elf2cfetbl'...
Cloning into '/home/jp/cFS/test/tools/tblCRCTool'...
Submodule path 'apps/ci_lab': checked out 'e4a1bb83357c25fac89a5c7fc240a925ecd66094'
Submodule path 'apps/sample_app': checked out '71d2c3290f4c18359cf9212346394e30b5d34fab'
Submodule path 'apps/sample_lib': checked out '14710ac0d777e40d5673d61b0fa6cc54c68200d3'
Submodule path 'apps/sch_lab': checked out '2e8c26797f2dbed50ad95b7b83b582b077753f9d'
Submodule path 'apps/to_lab': checked out '7e7dc4b5e10386ce533fdf890d7e2805cdece38c'
Submodule path 'cfe': checked out '84ec281b4a432db7f41f0814e37fc239c723ad39'
Submodule path 'osal': checked out '33471c7efa3d4e6a1e3760e0804c79fb21c3de54'
Submodule path 'psp': checked out '24450768d270eebbc887513a6e867942482ff730'
Submodule path 'tools/cFS-GroundSystem': checked out '9cd5304e749b0ca24e27f96ac8918dd1a613a65d'
Submodule path 'tools/elf2cfetbl': checked out '8f53aee4da49f2e432a1ef1396ba5e2df885e733'
Submodule path 'tools/tblCRCTool': checked out 'c8009d3fa9cc397f1b989df0bac71c6830097a65'
XXX@XXX~/cFS/test$ cp cfe/cmake/Makefile.sample Makefile
XXX@XXX~/cFS/test$ cp -r cfe/cmake/sample_defs/ sample_defs
[... 272 lines cut ...]
Install the project...
-- Install configuration: "debug"
-- Installing: /home/jp/cFS/test/build/exe/host/cmdUtil
-- Installing: /home/jp/cFS/test/build/exe/host/elf2cfetbl
-- Installing: /home/jp/cFS/test/build/exe/host/cfe_ts_crc
Built target mission-install
XXX@XXX~/cFS/test$ cd build/exe/cpu1/
XXX@XXX~/cFS/test/build/exe/cpu1$ ./core-cpu1 
OS_BSP_Initialize():Maximum user msg queue depth = 10
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
OS_Posix_GetSchedulerParams():188:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():188:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():374:Selected policy 2 for RT tasks, root task = 99
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
OS_Posix_InternalTaskCreate_Impl():473:pthread_attr_setstacksize error in OS_TaskCreate: Invalid argument
Aborted (core dumped)
```
</details>

**System observed on:**
 - VirtualBox 6.1 running Linux Mint 20, Kernel: Linux 5.4.0-40-generic, Architecture: x86-64

skliper, 2020-07-15:
Quite possibly due to one of the known bugs addressed in the next round of integration candidates (the stack size minimum/rounding update).

jphickey, 2020-07-15:
Concur - this is a recent issue fixed in nasa/osal#528 but hasn't made its way into "master" branch just yet.  If you'd like to get the fix you could pull the "integration-candidate" branch instead, it is merged into that one.  Just be aware that that branch may still change before the final merge.

mentoc3000, 2020-07-16:
Switching to `integration-candidate` on `osal` does fix the issue for me. Thanks! I'll keep an eye out for when it's merged into `master`.