nasa/cFS #113: No tests were found

fletching, 2020-07-17:
**Describe the bug**
`make test` on a freshly cloned repository does not find any tests.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/nasa/cFS.git`
2. `cd cFS`
2. `cp cfe/cmake/Makefile.sample Makefile`
3. `cp -r cfe/cmake/sample_defs sample_defs1
4. `make SIMULATION=native ENABLE_UNIT_TEST=true prep`
5. `make`
6. `make install`
7. `make test`

**Observed behavior**

```
lcov --capture --initial --directory build/native/default_cpu1 --output-file build/native/default_cpu1/coverage_base.info
Capturing coverage data from build/native/default_cpu1
Found gcov version: 9.3.0
Using intermediate gcov format
Scanning build/native/default_cpu1 for .gcno files ...
geninfo: WARNING: no .gcno files found in build/native/default_cpu1 - skipping!
Finished .info-file creation
(cd build/native/default_cpu1 && ctest -O ctest.log)
Test project /home/XXX/cFS/cFS-ic/build/native/default_cpu1
No tests were found!!!
```

**Expected behavior**
Tests run and pass.

**System observed on:**
VirtualBox 6.1 running Linux Mint 20, Kernel: Linux 5.4.0-40-generic, Architecture: x86-64

jphickey, 2020-07-17:
Note - something like this could be caused by submodules being out of sync.  Can you confirm if you did a "git submodule update'  as part of this?

There was a recent change that affected the directory structure of the build tree, so if this was out of sync it would potentially manifest as not finding any test files.

fletching, 2020-07-17:
Yes, I forgot to include the `git submodule init` and `git submodule update` commands from my steps to reproduce. I did run `git submodule update` as part of my setup and got those results. I tried it both on `master` and `integration-candidate` and the same thing happened.

jphickey, 2020-07-17:
Did you do this before or after the "cp cfe/cmake/Makefile.sample Makefile" step?

The sample makefile might contain a line "ARCH ?= native" or "ARCH ?= native/default_cpu1" reflecting the recent change to the build tree.

Check your generated build tree and see if the actual build binaries are in the "native" subdir or a "native/default_cpu1" subdir.

fletching, 2020-07-17:
I updated the submodules prior to copying the Makefile, even after changing to the `integration-candidate` branch.

The sample Makefile I copied has `ARCH ?= native` on `master` and `ARCH ?= native/default_cpu1` on `integration-candidate`. The directories for the binaries are correct for both branches.

jphickey, 2020-07-17:
Ah hah --- I see it -- `ENABLE_UNIT_TEST=true` should be `ENABLE_UNIT_TESTS=true`

fletching, 2020-07-17:
:facepalm: Yes! Thank you!!

closed 2020-07-17 by fletching