nasa/cFS #121: How can I include a prebuilt library as a CFS library?

stephen99scott, 2020-08-10:
Hi, I would like to include the [CubeSat Space Protocol Library](https://github.com/libcsp/libcsp) (CSP) as a library in CFS.

I am new to CFS so I am unsure the correct way to approach this. 

My goal is to use the CSP functionality within the CI/TO applications in order to send to/receive from CAN. I would like to use the waf build system included with CSP to build the library, then link the output library to CFS so the CI/TO apps can use the CSP functionality by including the appropriate headers. As a prereq I have added the libcsp repo to `apps/libcsp`.

It is my understanding that adding libcsp to my `targets.cmake` `TGT1_APPLIST` is not ideal, because that would attempt to build the libcsp code; however, in this case it would have already been built using the waf build system.

I have tried the following:
1. Building libcsp as a shared library using the waf build system
1. Adding an include directive for the CSP headers in the TO application
1. Adding `include_directories(path_to_csp)` in `to/CMakeLists.txt`
1. Copying the waf-built `libcsp.so` to the target install directory `build/exe/cpu1/cf/libcsp.so`
1. Adding `cf/libcsp.so` to my target startup script

As a result, the library fails to load at startup. I also tried:
1. Building libcsp as a shared library using the waf build system
1. Adding an include directive for the CSP headers in the TO application
1. Adding `include_directories(path_to_csp)` in `to/CMakeLists.txt`
1. Adding `target_link_libraries(core-${TGTNAME} path_to_csp/libcsp.so)` to `cfe/cmake/target/CMakeLists.txt` before the final executable is installed

As a result, the TO application fails to load.

Any help would be appreciated. I am unsure of the feasibility of using two separate build systems to build the code for my target, although I would like to avoid having to convert the waf build script to a cmake build if possible.

Thanks for your help.
Stephen

jphickey, 2020-08-10:
Your approach seems OK...  Can you clarify what you mean by:

> As a result, the library fails to load at startup

Is there a particular error?  What version of CFE are you running?  Is it the latest RC (6.8.0-rc1)?  Also what are you putting in `cfe_es_startup.scr`, in particular for the entry point?

stephen99scott, 2020-08-10:
I am using cFE 6.7.19.

This is the error given when attempting to load libcsp.so.
[image]
Here is the startup script entry.
[image]
This is the entry point function which I defined in `libcsp/src/csp_init.c`
[image]
Here is the error given for the to_lab as well.
[image]

jphickey, 2020-08-10:
Looks like the startup script entry is missing a comma between the filename and entry point?  That's probably why it is saying invalid ES startup file entry.  It's not even trying to load the library.

Couple other items to consider:
- You should enable `OSAL_CONFIG_DEBUG_PRINTF` in your osconfig.cmake file.  In the event of module loading issues, it will give much more useful information this way, not just "OS_ERROR".
- You should also be able to specify the library entry point as NULL in your startup script if you don't actually have/need one.  This is intended for 3rd party libraries that don't have a CFE entry point.
- You may also use `target_link_libraries()` (your second approach) but you must put this in the _app_ that uses it (e.g. in TO, not in CFE).  Be aware that this probably only works on Linux though - or a system that otherwise has a fully-functional shared library runtime loader.  RTOS's like RTEMS and VxWorks generally do not have this capability.

stephen99scott, 2020-08-10:
> Looks like the startup script entry is missing a comma between the filename and entry point? That's probably why it is saying invalid ES startup file entry. It's not even trying to load the library.

Good catch - sometimes a second pair of eyes is all it takes. The module loads successfully now.

> You should enable OSAL_CONFIG_DEBUG_PRINTF in your osconfig.cmake file

Sounds good, I'll enable this for future troubleshooting.

> You should also be able to specify the library entry point as NULL in your startup script if you don't actually have/need one

I will do this as well.

> Be aware that this probably only works on Linux though - or a system that otherwise has a fully-functional shared library runtime loader

My team is intending to port CFS to freeRTOS, so it would probably be best to avoid that method then.

Thanks for all your help!

closed 2020-08-10 by stephen99scott