nasa/cFS #150: Sample "real-life" implementation of a cFS mission build.

astrogeco, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
From the cFS Mailing list:

>it would be nice to see a fully completed project to decipher why certain features are used. The how is easy to digest from the documentation but a real-world example would help clarify each feature’s intended purpose.
>
>I know a lot of this code may be internal to NASA / other organizations but it would be really helpful to have a completed demo to reference from time to time. 

**Describe the solution you'd like**
A how-to guide or a complete docker image with a mission-like cFS implementation.

**Describe alternatives you've considered**
The current bundle, partnering with open-sat kit

**Additional context**
None

**Requester Info**
@astrogeco

skliper, 2020-10-15:
I'd be interested to hear how this would differ from the open-sat kit.

jeffstjean, 2020-10-19:
Hi all!

This started from a small discussion a couple of us were having in the cFS mailing list. I'm definitely not at a state where I feel as though I've exhausted all (or even most) of the documentation but there were a couple things I felt could help the learning process. Maybe some of this already exists and I just haven't looked hard enough but hopefully this will point the community in the right direction.

As a particular example, I was following the OpenSatKit tutorials on YouTube after completing the slide deck exercises from April. I got to the section about tables and understood _exactly_ how to interface with the cFE and implement my own tables. Then I realized, I didn't really know _why_ I was doing what I did. Are tables an in-memory store for various configurations/variables? Or are they more suited for holding the "current" data for a particular app (maybe say the current IMU readings)? While this may seem like a fairly basic question in retrospect, it made me realize that I didn't have a grasp on the _why_ for some of the features in cFS. 

Again, I'm still fairly new to the community and still have a lot of ground to cover. maybe there's some pre-existing way to discover these sorts of answers, but, as a few of us were commenting, it might be nice to see a fully-completed project (similar to OpenSatKit) that includes both development and production environments and documentation on the "typical" cFS setup. It might be nice for beginners to see the difference between the local ip command ingest and how you'd transition that to interface with actual hardware while properly abstracting as necessary.

Thoughts?