nasa/cFS #155: Help: Could not load cFE application file

moafandi0, 2020-11-03:
Hi,

I'm working on an app that uses a library for an IMU. My app compiles without errors but when I run the core-cpu1, I get:
`Could not load cFE application file:/cf/imu.so. EC = 0xFFFFFFFF`

My app directory is called imu and it contains:
/imu
---fsw
---imu_lib
------build
---------obj
------------vn ( which includes a bunch of .o files)
---------libvnc.a (contains the .o files)
------include
---------vn (which includes a bunch of .h files)
---CMakeLists.txt

Here is the CMakeLists.txt

`
cmake_minimum_required(VERSION 2.6.4)
project(CFS_IMU C)

include_directories(fsw/mission_inc)
include_directories(fsw/platform_inc)
include_directories(fsw/src)

include_directories(imu_lib/include)
include_directories(imu_lib/include/vn)

aux_source_directory(fsw/src APP_SRC_FILES)
add_cfe_app(imu ${APP_SRC_FILES})
target_link_libraries(imu /home/eaglecam/CFS-101.initial/apps/imu/imu_lib/build/libvnc.a)
`

Any help is appreciated.

Thanks

jphickey, 2020-11-03:
Failing to load a dynamic shared object is (almost) always related to symbol resolution.

First be sure you have enabled the OSAL debug statements, i.e. `set(OSAL_CONFIG_DEBUG_PRINTF TRUE)` in your osconfig.cmake file.  At least on Linux this will tell you what symbol is the issue.

Also note when linking a static lib (.a) into a shared object (.so) you'll need to set the POSITION_INDEPENDENT_CODE property to be true on the static lib.  By default it is false on static libs.  See https://cmake.org/cmake/help/v3.0/prop_tgt/POSITION_INDEPENDENT_CODE.html

I've also found it helpful to use the `nm` tool to list the objects in the resulting `.so` file to be sure it has all the symbols I expect it to.

moafandi0, 2020-11-04:
I recompiled the library libvnc.a with the `-fPIC` flag in make and it worked! Thanks!

closed 2020-11-04 by moafandi0