nasa/cFS #157: can't run unit tests

smalltalker55, 2020-11-13:
**Describe the bug**
I can build and run the application, but when I try to run unit tests I am told "no tests were found".  Probably I am not following instructions properly, some parts seem a little vague to me.

**To Reproduce**
After setup, I followed the instructions in "Build and Run" which were

make SIMULATION=native prep
make
make install
cd build/exe/cpu1/
./core-cpu1

and I got results that looked reasonable.   The instructions then said "Unit tests can be added with ENABLE_UNIT_TESTS=true, run with make test, and coverage reported with make lcov."

So, I tried "make SIMULATION=native ENABLE_UNIT_TESTS=true prep" followed by 

make
make test

and also just "make ENABLE_UNIT_TESTS=true prep" followed by 

make
make test

I guess the instructions are not clear to me.   (I tried another few combinations, too, but these seemed the most reasonable).
**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: [e.g. Linux 4.4]
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

smalltalker55, 2020-11-13:
Sorry, didn't edit the whole template.

This is on Ubuntu 16.04

I'm not that familiar with cmake though I am usually good at following instructions.   Not this time, though.

-Ralph Johnson

jphickey, 2020-11-13:
The `ENABLE_UNIT_TESTS` option is cached by CMake.   Once initially set these keep their original value from run to run.  To clear/reset the cache variable you need to fully clean your build.  This can be done via `make distclean` or just by removing your `build` directory entirely.

Caching is intentional so you only have to specify what you want on the first time you run "prep" and initially generate the build tree.  From thereon it remembers the relevant settings and continues to use them, making subsequent invocations simple.  While the option value _can_ technically be changed using a tool like cmake-gui or by editing the cache file, the easiest way is to simply remove your entire build tree and start clean again.  

The same paradigm is true for many other build options (cross compiler/SIMULATION, BUILDTYPE, etc).

smalltalker55, 2020-11-14:
Thanks!    That makes everything work better.

closed 2021-01-04 by skliper