nasa/cFS #164: why I could not update cFS

aogrcs, 2020-12-21:
Dear,
I clone the repository using git clone, cd to the directory then git submodule init, but when update it, it could not get the cfe, is there any wrong? Thanks
I can update the cFE repository.
Sincerely

skliper, 2020-12-21:
Worked for me.  Please provide the exact steps to replicate your issue.

Working sequence:
```
> git clone git@github.com:nasa/cFS.git
Cloning into 'cFS'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 605 (delta 1), reused 4 (delta 1), pack-reused 599
Receiving objects: 100% (605/605), 3.80 MiB | 13.76 MiB/s, done.
Resolving deltas: 100% (275/275), done.
> cd cFS
> git submodule update --init
Submodule 'apps/ci_lab' (https://github.com/nasa/ci_lab.git) registered for path 'apps/ci_lab'
Submodule 'apps/sample_app' (https://github.com/nasa/sample_app.git) registered for path 'apps/sample_app'
Submodule 'apps/sample_lib' (https://github.com/nasa/sample_lib.git) registered for path 'apps/sample_lib'
Submodule 'apps/sch_lab' (https://github.com/nasa/sch_lab.git) registered for path 'apps/sch_lab'
Submodule 'apps/to_lab' (https://github.com/nasa/to_lab.git) registered for path 'apps/to_lab'
Submodule 'cfe' (https://github.com/nasa/cFE.git) registered for path 'cfe'
Submodule 'osal' (https://github.com/nasa/osal.git) registered for path 'osal'
Submodule 'psp' (https://github.com/nasa/PSP.git) registered for path 'psp'
Submodule 'tools/cFS-GroundSystem' (https://github.com/nasa/cFS-GroundSystem.git) registered for path 'tools/cFS-GroundSystem'
Submodule 'tools/elf2cfetbl' (https://github.com/nasa/elf2cfetbl.git) registered for path 'tools/elf2cfetbl'
Submodule 'tools/tblCRCTool' (https://github.com/nasa/tblCRCTool.git) registered for path 'tools/tblCRCTool'
Cloning into '/export/home/jhageman/cFS/cFS/apps/ci_lab'...
Cloning into '/export/home/jhageman/cFS/cFS/apps/sample_app'...
Cloning into '/export/home/jhageman/cFS/cFS/apps/sample_lib'...
Cloning into '/export/home/jhageman/cFS/cFS/apps/sch_lab'...
Cloning into '/export/home/jhageman/cFS/cFS/apps/to_lab'...
Cloning into '/export/home/jhageman/cFS/cFS/cfe'...
Cloning into '/export/home/jhageman/cFS/cFS/osal'...
Cloning into '/export/home/jhageman/cFS/cFS/psp'...
Cloning into '/export/home/jhageman/cFS/cFS/tools/cFS-GroundSystem'...
Cloning into '/export/home/jhageman/cFS/cFS/tools/elf2cfetbl'...
Cloning into '/export/home/jhageman/cFS/cFS/tools/tblCRCTool'...
Submodule path 'apps/ci_lab': checked out 'adc0edb707bdd8595654676deddf4dda4f39c669'
Submodule path 'apps/sample_app': checked out 'b798efc39900574fa2b08c2588fa2c6e2254ec99'
Submodule path 'apps/sample_lib': checked out '8f81bfdfda3ace8fc9f9f18d4e8bd7ec3b7ef4b9'
Submodule path 'apps/sch_lab': checked out 'f7b0595e9184cf5b846dc405a18deaac471f8ec8'
Submodule path 'apps/to_lab': checked out '47793f0c9d74bde8b3779717b779ff3c955d7eea'
Submodule path 'cfe': checked out '56397a3a2c47c674892c14610e78263030441d73'
Submodule path 'osal': checked out 'd698a4d7ddc66826a4c7c287468b5f1aa6ca372f'
Submodule path 'psp': checked out '317892063e94a063337a6713eb7a24ab69d2ed69'
Submodule path 'tools/cFS-GroundSystem': checked out '10efa7050f82a4acc3013e9fddf31fd0ea5efe28'
Submodule path 'tools/elf2cfetbl': checked out 'eb5e326a3c85af2dd34735913881f8e15db808db'
Submodule path 'tools/tblCRCTool': checked out 'f21a641c784b49a811e548b71322b953332e53a3'
```

aogrcs, 2020-12-22:
It worked, maybe the unstable wireless network caused this, thanks

closed 2020-12-22 by aogrcs