nasa/cFS #185: Apply CodeQL to Integration Candidate branch

ArielSAdamsNASA, 2021-02-03:
**Describe the bug**  
The integration candidate branch is not scanned by CodeQL on push. 

**Expected behavior**
Scan the integration-candidate branch on push using CodeQL.

**Additional context**
Can either simply add the integration-candidate branch or every branch can be added using '**'. 

**Reporter Info**
Ariel Adams, ASRC Federal

skliper, 2021-02-04:
I recommend scanning on every branch (push and pull-request) for now.

astrogeco, 2021-02-05:
One reason the original templates had 
```yml
push: 
-main 
pull_request
```

is that for the integration-candidate PRs everything was running twice. I don't think that's a big deal for thei workflow though so I'm fine keeping it as 

```yml
on:
  push:
  pull_request:
```

The other idea is to exclude the integration-candidate from the push trigger but that seems overly complicated

skliper, 2021-02-07:
Actually, I wonder if that warning was just coming up because an initial run on main wasn't recorded yet... I bet the first option might actually be OK.  Note I tried both on my CodeQL addition to ci_lab and they both gave me the warning (I think because it hasn't been run on main yet).

ArielSAdamsNASA, 2021-02-08:
> Actually, I wonder if that warning was just coming up because an initial run on main wasn't recorded yet... **I bet the first option might actually be OK.** Note I tried both on my CodeQL addition to ci_lab and they both gave me the warning (I think because it hasn't been run on main yet).

The first option being the workflow only using the main branch on push? 

I do not see the warning when the main branch is ran on both push and pull-request which is how it is set up currently. However, the error was still occurring multiple times before the main branch was set on both push and pull-request (only set on push). I do still want to make some minor changes to the CodeQL workflow that resembles @skliper's ci_lab workflow which includes: 
- Setting timeout
- Removing the run cFS stage
- Removing the configuration file and adding the queries to the workflow directly 

Latest ci_lab CodeQL run (main branch on push and pull-request):
[image]

Latest cFS CodeQL run (main branch on push and pull-request):
[image]

astrogeco, 2021-02-08:
What is the downside of having the separate configuration file? It might be useful in case we wanted to use CodeQL locally: https://codeql.github.com/docs/codeql-cli/

ArielSAdamsNASA, 2021-02-08:
> What is the downside of having the separate configuration file? It might be useful in case we wanted to use CodeQL locally: https://codeql.github.com/docs/codeql-cli/

@astrogeco There's not really a downside of having a separate configuration file besides the project having an extra folder and file. I can keep the configuration file. So, the ci_lab should also use a configuration file for its CodeQL workflow?

skliper, 2021-02-09:
@astrogeco from what I've seen from the CLI, everything we need can be specified on the command line.  The only thing that was in the configuration file was the queries (just the default two), which can be specified in the workflow or on the command line without maintaining an extra file.  I see it as just another level of indirection that doesn't really provide much benefit.  That said, it's trivial to change either way if someone wants to add CodeQL CLI makefile rules or whatever so I'd suggest deferring decisions such as this to when it actually gets implemented.  Things may change between now and then, so excessive thrash at this point for something that may or may not happen at some point doesn't seem to provide much value.  If we start adding custom queries to cover our coding standards it may be valuable to use a standard config file, but that's in the same boat as future work.

ArielSAdamsNASA, 2021-02-11:
FYI, I noticed that if you add `repository: nasa/cFS` in the checkout bundle, you can just put the main branch on push without any warnings or errors.

PR #187 (2021-02-16): Fix #185, Apply CodeQL to Every Branch
**Describe the contribution**
Fix #185 Removed main branch on push and pull-request. Added a timeout. Removed cFS running. 

**Expected behavior changes**
CodeQL should run on every branch of cFS and timeout. cFS should not run, only build, since CodeQL does not require cFS to run for analysis. 

**System(s) tested on**
 CI

**Contributor Info - All information REQUIRED for consideration of pull request**
Ariel Adams, ASRC Federal
commit a8fde080 (2021-02-16): Fix #185, Apply CodeQL to Every Branch
commit fd40b9e7 (2021-02-16): Merge pull request #187 from ArielSAdamsNASA/CodeQLOn

Fix #185, Apply CodeQL to Every Branch
closed 2021-02-16 by astrogeco