nasa/cFS #188: Question: How do I ensure that the ground system sends a command with the correct secondary header?

JackNWhite, 2021-02-12:
I have copied my first test app from the sample app in the bundle. I have now been able to add my commands, but `CFE_MSG_GetFcnCode()` always returns `CFE_MSG_WRONG_MSG_TYPE` because `hassechdr == false`.

For the sample app commands, `hassechdr == true`, but of course those commands were already pickled, so I don't know where the problem lies.

Does the ground station or the `CHeaderParser.py` script add the headers automatically? How is this worked out? I have been just adding three `uint64`s to my command.

skliper, 2021-02-12:
@JackNWhite Note that depending on your selected implementation, the bits in MsgId that indicates the presence of the secondary header and if a message is cmd or tlm.  If you are using the default, MsgID maps directly to the CCSDS Stream ID, which is defined here:
https://github.com/nasa/cFE/blob/fcaa1d0c6b9e8244b071b8cafc765078680e24b4/fsw/cfe-core/src/inc/ccsds_hdr.h#L54-L59

So for your new commands, you need to pick a MsgId that has the appropriate bits set (0x1800 for cmd, 0x0800 for tlm).

See also https://github.com/nasa/cFE/issues/590, which highlights documentation deficiencies.

JackNWhite, 2021-02-15:
Thank you muchly.

closed 2021-02-16 by skliper