nasa/cFS #189: EVS Port1 42/1/CFE_ES 48: ES_ProcControlReq: Unknown State ( 1 ) Application IMU.

moafandi0, 2021-02-14:
Hi,
I'm working on an IMU application that connects and logs IMU data.
I have a function `IMU_InitSensor() ` that initializes the sensor and returns CFE_SUCCESS if initialization succeeds and -1 otherwise. if the return value is -1, the App will exit.
I'm trying to run the app when the sensor is physically disconnected to make sure I get the correct behavior, but I'm getting this error message:
`
1980-012-14:03:21.76253 Application IMU called CFE_ES_ExitApp

EVS Port1 42/1/CFE_ES 48: ES_ProcControlReq: Unknown State ( 1 ) Application IMU.
`
Here is my AppMain:

``
void IMU_AppMain()
{
    int32 iStatus = CFE_SUCCESS;
    uint32 taskId = 0;

    
`
    /* Start Performance Log entry */
    //CFE_ES_PerfLogEntry(IMU_MAIN_TASK_PERF_ID);

    /* Perform application initializations */
    if (IMU_InitApp() != CFE_SUCCESS)
    {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_EXIT;        
        goto IMU_AppMain_Exit_Tag;
    }
    else if (IMU_InitSensor() != CFE_SUCCESS)//returns 0 on success and -1  #otherwise
    {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_EXIT;        
        goto IMU_AppMain_Exit_Tag;
    } 
    else {
        /* Do not perform performance monitoring on startup sync */
        CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);
        CFE_ES_WaitForStartupSync(IMU_TIMEOUT_MSEC);
        CFE_ES_PerfLogEntry(IMU_MAIN_TASK_PERF_ID);
    }

    
    
    iStatus = CFE_ES_CreateChildTask(&taskId,
                                     "IMU Child Task",
                                     IMU_ChildTask,
                                     IMU_CHILD_TASK_STACK_PTR, 
                                     IMU_CHILD_TASK_STACK_SIZE, 
                                     IMU_CHILD_TASK_PRIO,
                                     0);
    if(iStatus != CFE_SUCCESS)                                     
    {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_ERROR;
        CFE_ES_WriteToSysLog("IMU - Failed to create child task\n");
        goto IMU_AppMain_Exit_Tag;

    }

    /* Application main loop */
    while (CFE_ES_RunLoop(&g_IMU_AppData.uiRunStatus) == TRUE)
    {
        CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);

        iStatus = IMU_RcvMsg(CFE_SB_POLL);

    }    

    

    /* Stop Performance Log entry */
    CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);
`
IMU_AppMain_Exit_Tag:
    /* Exit the application */
    CFE_ES_WriteToSysLog("IMU - Exiting App...\n");
    CFE_ES_ExitApp(g_IMU_AppData.uiRunStatus);
} 
`

Any help? Thanks!

dmccomas, 2021-02-15:
Do you set g_IMU_AppData.uiRunStatus = CFE_ES_APP_RUN before you enter the
main loop?    This needs to be done. For an example see  QQ_AppInit() in
the cFE app's developer's guide this is done.

Hope this helps - Dave

jphickey, 2021-02-15:
Is it just the single "unknown state" log message that is the issue here?

I noticed this extra log message when testing another unrelated item, it was just an artifact of the app exiting itself and not really indicative of a problem.  Perhaps it should be cleaned up not to say "unknown state" which sounds bad but its really not.

Does the app proceed to exit and the system work normally thereafter?  (it should).

moafandi0, 2021-02-15:
> Do you set g_IMU_AppData.uiRunStatus = CFE_ES_APP_RUN before you enter the main loop? This needs to be done. For an example see QQ_AppInit() in the cFE app's developer's guide this is done. Hope this helps - Dave
> […](#)

Thanks for your reply!

This is the first thing done inside IMU_InitApp()

> Is it just the single "unknown state" log message that is the issue here?
>
> I noticed this extra log message when testing another unrelated item, it was just an artifact of the app exiting itself and [...]

The unknown state message is printed to the terminal every 1 second but the system works normally. However, I don't think the app is exiting because in other cases, when I exit the app I get the following:

`1980-012-14:04:20.05614 Application IMU called CFE_ES_ExitApp
EVS Port1 42/1/CFE_ES 13: Exit Application IMU Completed.
`

I don't get the "Exit application completed" here in this case.

jphickey, 2021-02-15:
You didn't mention what version of CFE you are running...  is this the "main" branch or a tagged release?

There was an old bug, see nasa/cFE#89.

moafandi0, 2021-02-16:
> You didn't mention what version of CFE you are running... is this the "main" branch or a tagged release?
> 
> There was an old bug, see [nasa/cFE#89](https://github.com/nasa/cFE/issues/89).

I'm using version 6.5.0 that I got from sourcefourge and it looks like this is exactly the issue I'm having.

As I understood from the link you sent, I tried to set the runStatus to CFE_ES_SYS_DELETE like this:

```
else if (IMU_InitSensor() != CFE_SUCCESS)//returns 0 on success and -1  #otherwise
{
    g_IMU_AppData.uiRunStatus = CFE_ES_SYS_DELETE;        
    goto IMU_AppMain_Exit_Tag;
} 
```

 But the issue still persists. I'm sure there's something else here I missed.

jphickey, 2021-02-16:
>  I'm sure there's something else here I missed.

Probably not, actually....  There is just a bug in that version that ultimately means apps cannot self-exit -- they get stuck in a loop and never cleans up.  It was fixed a while back (see the ticket) but will not be back-ported to 6.5.0 unless someone in the community chooses to fix it in that version.

Unless there is a good reason you need to stick to an old version, for any new work I would recommend using a more current cFE version.  Almost all of the 1000+ reports in the CFE repo are probably other issues you may run into.

moafandi0, 2021-02-18:
In the end of the child task that I spawn, I set `g_IMU_AppData.uiRunStatus = CFE_ES_APP_EXIT` and then exit the child task. In that case, the app exits normally. Is it because the app received this commands while it is in its processing loop? 

I will see if I can update it but I wanna try finding a workaround first.

moafandi0, 2021-02-19:
Here's a workaround, Instead of using a goto statement, I let the the app continue its normal path until it gets to the exit app, and I enclosed the createChildTask function in an if statement to avoid creating it if SensorInit() fails and now I don't get the error msg and the app exits normally. See code below. 
Thanks for your help everyone!

```
    if (IMU_InitApp() != CFE_SUCCESS)
    {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_EXIT;        
        //goto IMU_AppMain_Exit_Tag;
    }
    else if (IMU_InitSensor() != CFE_SUCCESS)
    {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_EXIT;        
        //goto IMU_AppMain_Exit_Tag;
    } 
    else {
        /* Do not perform performance monitoring on startup sync */
        CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);
        CFE_ES_WaitForStartupSync(IMU_TIMEOUT_MSEC);
        CFE_ES_PerfLogEntry(IMU_MAIN_TASK_PERF_ID);
    }

    if(g_IMU_AppData.uiRunStatus == CFE_ES_APP_RUN)
    {
        iStatus = CFE_ES_CreateChildTask(&taskId,
                                     "IMU Child Task",
                                     IMU_ChildTask,
                                     IMU_CHILD_TASK_STACK_PTR, 
                                     IMU_CHILD_TASK_STACK_SIZE, 
                                     IMU_CHILD_TASK_PRIO,
                                     0);
        if(iStatus != CFE_SUCCESS)                                     
        {
        g_IMU_AppData.uiRunStatus = CFE_ES_APP_ERROR;
        CFE_ES_WriteToSysLog("IMU - Failed to create child task\n");
        //goto IMU_AppMain_Exit_Tag;

        }

    }
    
    /* Application main loop */
    while (CFE_ES_RunLoop(&g_IMU_AppData.uiRunStatus) == TRUE)
    {
        CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);

        iStatus = IMU_RcvMsg(CFE_SB_POLL);

    }    

    /* Stop Performance Log entry */
    CFE_ES_PerfLogExit(IMU_MAIN_TASK_PERF_ID);

IMU_AppMain_Exit_Tag:
    /* Exit the application */
    CFE_ES_WriteToSysLog("IMU - Exiting App...\n");
    CFE_ES_ExitApp(g_IMU_AppData.uiRunStatus);
}
```

commit d6d4d652 (2021-09-01): nasa/cFS-GroundSystem#190, Fix #189, Virtualenv and Pipenv .gitignore support

nasa/cFS-GroundSystem#194, Fix doc, comment, and message typos
commit 4736ea66 (2021-09-01): IC:2021-08-31, cFE, osal, cFS-GroundSystem

**Combines**

nasa/cFE#1885,              v6.8.0-rc1+dev980
nasa/osal#1138,             v5.1.0-rc1+dev598
nasa/cFS-GroundSystem#195,  v2.2.0-rc1+dev63

**Includes**

*cFE*

nasa/cFE#1870, Add SB API test cases
nasa/cFE#1869, Add ES API test cases
nasa/cFE#1872, Add TBL API test cases
nasa/cFE#1871, Add FS API test cases
nasa/cFE#1860, Add Time Clock Test
nasa/cFE#1862, EVS coverage test
nasa/cFE#1876, SB test improvements
nasa/cFE#1865, CFE_TBL_Modified: Test CRC, updated flag
nasa/cFE#1881, Improve EVS code coverage
nasa/cFE#1877, add call to CFE_ES [... cut]
closed 2021-02-19 by moafandi0