nasa/cFS #191: Question: Getting cFS to communicate with Ground Station running on different host

RyanDozier, 2021-02-15:
Hey,

So this may have a simple solution, but I'm new to cFS and I'm trying to figure out how to get a Ground Station running on a VM to communicate with a cFS instance running on a Raspberry Pi to communicate.  I can see the packets being received by the cFS instance but it doesn't send any data.  There are no firewalls blocking anything. I'm also running the sample Makefile and sample_defs. 

I know I need to modify the ci_lab and to_lab files, but I'm not sure what to modify.

Ryan

skliper, 2021-02-16:
> I can see packets being received...

Do you mean if you send "ES No-Op" from cFS-GroundSystem (via clicking the button on the command subsystem) , you see response in cFS? Something like:
```
EVS Port1 66/1/CFE_ES 3: No-op command:
 cFS Versions: cfe v6.8.0-rc1+dev290, osal v5.1.0-rc1+dev221, psp v1.5.0-rc158
```

If so, have you tried sending the "Enable Tlm" command, putting in the IP address of the VM for the dest_IP Input?

RyanDozier, 2021-02-18:
I've tried putting the IP address into the Enable Tlm and I see it send the UDP packet to the other host.  I didn't check to see if the cFS sends any packets to the 127.0.0.1, but I also don't see any response from the cFS.  So I didn't think to look.

The Ground station and the cFS don't share the same host and have different IPs.  The cFS and ground station communicate just fine when they are co-located.  But I need to get them to communicate across the network for my purposes.

Any help would be greatly appreciated and I also appreciate the response.

skliper, 2021-02-18:
cFS will only send packets to the address you enable via the "Enable Tlm" command.  As long as the host is configured to send to the IP of the target (commands) from within cFS-GroundSystem, and cFS is configured to send to the IP of the VM things should work.  If they don't, it could be a firewall or other network issue.

RyanDozier, 2021-02-19:
Hey,

Thanks that definitly helped.  I got it to work.  Is there a place I can ask other questions if need be?  I don't want to spam the issues section of github with random questions.

Thanks!!

skliper, 2021-02-19:
@RyanDozier I suggest joining the community email list via the link on the cFS README.md, it's a very helpful group in general, and easy to splinter from for detailed topics not of interest to the general group.

closed 2021-02-22 by RyanDozier