nasa/cFS #192: Check that commit messages adhere to standard using the continuous integration workflow

astrogeco, 2021-02-16:
**Is your feature request related to a problem? Please describe.**
Community contributors are not familiar with our `Fix #XYZ, SHORT_DESCRIPTION` convention for commit messages. 

**Describe the solution you'd like**
Adding an extra step or job to "format check" workflow

**Describe alternatives you've considered**
Add a pre-commit hook on git

**Additional context**
None

**Requester Info**
@astrogeco

chillfig, 2022-04-22:
@astrogeco Would you like "**HotFix** #XYZ, <SHORT_DESCRIPTION>" to also pass for future commit messages? Such as in 06f5ce8eabddf7a3005242f55019f9a6daa9c939

skliper, 2022-04-22:
We need to also support the optional long description...
```
[Fix|Hotfix] #XXX, SHORT_DESCRIPTION

<LONG_DESCRIPTION>
```

astrogeco, 2022-04-22:
We also use "Partial #XYZ". 

So maybe have three "acceptable" keywords: 
- Fix
- HotFix
- Partial

astrogeco, 2022-04-22:
thanks for tackling this!

chillfig, 2022-04-25:
Thank you! Yes, I am just requiring the first line and the rest is free-form.

commit 7bda14ac (2022-05-19): Merge pull request #470 from chillfig/commit_msg_check

Fix #192, Adds check commit message job in format check
commit efbb7295 (2022-05-19): Fix #192, Adds check commit message job in format check

Adds pattern that accepts commit msgs with "Fix [OR] HotFix [OR] Partial #XYZ <description>",
Adds pattern that accepts commit msgs with "Merge pull request #XYZ <description>",
Adds pattern that accepts commit msgs with "IC: <description>"
Adds conditional for commit msg job to run only on pull-requests
Splits msg check for PR title and each of its commits
Adds contextual comments to each step
Adds implicit default flags to each step for readability