nasa/cFS #194: Create Contributing.md

ArielSAdamsNASA, 2021-02-19:
**Is your feature request related to a problem? Please describe.**
Create a Contributing.md file so users know how to engage with cFS. 

**Describe the solution you'd like**
Include instructions on how to report bugs, request a feature, participate in discussions, create pull requests, and view security vulnerabilities. 

**Additional context**
I would like to create a code of conduct for this file. I would also like to explore how cFS could allow more discussions from the community such as using gitter.im, discourse, or a discussions template to use in issues.

**Requester Info**
Ariel Adams, ASRC Federal

skliper, 2021-02-19:
Good plan!  There was a draft started as a PR in cFE (I think), and I've sent Gerardo a few things on the PR submission process (and I think he's collected quite a bit of other info).  All good stuff to actually formalize and provide to users.  Could like to it from the submodules (vs writing one for every repo).

ArielSAdamsNASA, 2021-02-19:
@skliper I used the draft from cFE that you mentioned for the cFS contributing.md file, but there are some things missing and topics I would like to add. It would be great to get your feedback! Here is the [PR](https://github.com/nasa/cFS/pull/195)

commit 3fe84826 (2021-03-02): Fix #194, Create Contributing.md
commit 9b196f7b (2021-03-02): Fix #194, Create Contributing.md
commit fcf957e4 (2021-03-04): Fix #194, Create Contributing.md
commit 42cdfe80 (2021-03-04): Fix #194, Create Contributing.md
closed 2021-03-04 by astrogeco