nasa/cFS #203: Highlight community contributors

astrogeco, 2021-02-25:
**Is your feature request related to a problem? Please describe.**
We don't have a way of publicly recognizing our contributors from the community

**Describe the solution you'd like**
Add a contributors or community section to Readme.md with people's github icon

**Describe alternatives you've considered**

- Use https://github.com/all-contributors/all-contributors
- Use https://opencollective.com/
- Use https://remarkablemark.org/blog/2019/10/17/github-contributors-readme/
- Have a list on the Wiki with github handles

skliper, 2021-02-26:
Could be leveraged for #128

zachar1a, 2021-12-05:
Having #128 would be the coolest addition for community members imo.