nasa/cFS #207: Community space for Q&A and support

astrogeco, 2021-03-04:
Continuing the discussion about Questions and support started in #195.

astrogeco, 2021-03-04:
What would a "next-generation" community space need for us to be comfortable kicking questions "out" of github? I think the reasons discussed point towards that future.

zanzaben, 2021-03-04:
My biggest issue with Github is the quantity of stuff that currently goes through it. End users don't need to know about, for example, the 4 different issues in tblCRCTool that were made this week and all their pull requests. By that metric literally anything, even a different Github, would solve that main problem. 

It's important to have some way to tell people "Hey look at this, it's important" and not be the boy who cried wolf. Right now the email list does that because it isn't used that often, but if we get bigger and bigger, and anyone can send things to the email list, it will eventually have the same boy who cried wolf problem. 

This discussion thread is a perfect example of something that is important that lots of people should come look at.

jphickey, 2021-03-04:
While it is not "next-generation" (some might call it old-school) I actually do thing mailing lists are pretty good for this type of thing.  They've been around forever, and people know how to use them.  None of the so-called "next-generation" solutions (e.g. slack, zulip, rocket.chat) have the simplicity and ubiquity of a mailing list.  Everyone already has email.  They often don't want to (or cannot, in the case of many organizational policies) go through another 3rd party tool, which usually involves a separate user account, 3rd party software/clients etc.  In many ways that's a non-starter.

I think what is needed is just to find a long-term home for the mailing list that isn't dependent on NASA servers, allows users to subscribe/unsubscribe themselves easily, and offers a threaded archive service so they can search for old questions before posting new ones.

The next step is a wiki.  That is also old school by now, but newer than mailing lists.  It's a great place to put FAQs and deployment guides/HOWTOs/related info that isn't part of the source tree.  Any questions that occur on the mailing list more than once probably should get a FAQ entry to point to.

I agree both of my suggestions are neither new nor interesting, but sometimes the old way is indeed the best way.

astrogeco, 2021-03-04:
Some reference discussions form other groups that touch on some alternatives (you might be able to find my bias in these):
- [Discourse Use cases](https://blog.discourse.org/category/use-cases/)
- [Gitter VS Slack VS Discord (Unix RX project)](https://github.com/neuecc/UniRx/issues/327)
- [Discourse as a knowledge base](https://meta.discourse.org/t/using-discourse-as-a-internal-knowledge-base-intranet/33857/20)
- [Transition Google Groups to Slack/Gitter/Discourse](https://groups.google.com/g/php-fig/c/uN0EykvQkAA)
- [Gitter vs Slack vs Discord (flask project)](https://github.com/vimalloc/flask-jwt-extended/issues/283)
- [Replacing Slack by Discourse](https://meta.discourse.org/t/replacing-slack-by-discourse/121104/3)

skliper, 2021-03-04:
> The main objective is getting as many people as possible to interact with each other.

We have different main objectives, hence the reference to perspective.  I am not trying to get as many people as possible to interact nor am I trying to move everyone to GitHub, I just think it would be a step in the wrong direction to not allow questions to be submitted on GitHub until/unless there's a suitable replacement that's as easily accessible.  I'm advocating for a solution set that doesn't exclude portions of our community.

astrogeco, 2021-03-04:
Another good idea would be to have something like this from https://julialang.org/community/
[image]

astrogeco, 2021-03-04:
Similar concept form AdaCore https://www.adacore.com/community
[image]

ArielSAdamsNASA, 2021-03-04:
I like Discourse the most out of Slack, Gitter, Google Groups, Discord, Reddit, and Stack Overflow. Some noteworthy features includes GitHub integration, searchable discussions, and it uses forums for discussions rather than chat rooms. This allows for users to easily find posts since they can be organized into different categories and labelled with tags. It can also send notifications about new Discourse posts to ‘group chats’ on a number of instant messaging platforms. Another bonus is that it is free for open source projects. I feel like this would be a great place to have all community-based information such as Q&As, discussions, and announcements in one centralized application so GitHub can be used solely for developing cFS.

jphickey, 2021-03-05:
FWIW, this still seems like tossing up solutions in search of problems to me ...

Many times I've seen declarations to "lets go use product X" and it basically falls on deaf ears.  There was a push to go to a stack overflow channel a while back (that didn't go well) and also a slack channel (which also didn't go well, doesn't help that NASA has its own restrictions there).  At the end of the day everyone just continues to use the existing mailing list and the new tool of the day remains basically silent.  In the event that someone does try to use the new tool they don't get an answer because nobody is looking at it, and then they go back to the tried+true mailing list....

Perhaps with enough active encouragement we could get to an alternative discussion forum, but why?  Is it worth it?  Do we need it?  Do we have problems that are not addressed with a mailing list + github + wiki/faq solution?

Also I don't think there is a problem with letting users ask questions as github issues in the "cfs" repo.  Sometimes these can be answered outright (and we can close them) and other times they point out real bugs or enhancement needs - and they can be migrated to a real issue in another repo if that is the case, and you have the direct history.  Once again, I don't see anything broken or in need of changing there.

zanzaben, 2021-03-05:
I think an important distinction needs to be made on what the main goal of this new space is. Is it trying to get end users to engage with developers or have end users engage with other end users. Obviously both are good but there can be tradeoffs. For example the number of different platforms cFS is on. If it is on a bunch, Twitter, Discourse, Github, Reddit, email list, etc... cFS will reach the largest amount of people who have a way to get in contact with the devs, however it means the cFS team has to maintain all those different platforms, and it's possible to split the community across all the different platforms. On the other end of the spectrum there could be just 1 platform. It's easy for cFS to maintain and the entire community is together in one place, however there will be some people who for whatever reason don't/can't use that platform and they are lost. 

Or maybe the main goal is something completely different.

ArielSAdamsNASA, 2021-03-05:
Based off the feedback from the FSW Workshop, it seemed like the cFS community wanted a platform for discussions. So, in my opinion, the main goal of the community space is for users and developers to collaborate on ideas, ask questions, and chat about general topics. It would make things easier if we had the ability to use [GitHub discussions](https://docs.github.com/en/discussions/quickstart). 

>What is missing is a collaborative "strategic" process that involves the entire community. Roadmaps and a discussion on them are needed. GitHub issues are usually more "tactical" in nature?

zanzaben, 2021-03-05:
> FWIW, this still seems like tossing up solutions in search of problems to me ...

This is an example of one of the problems I see. https://github.com/nasa/cFS/issues/200#issuecomment-785927127. Developers asked for community input and a week later no one outside the dev team responded.  

But to your point maybe we can solve that by better using the tools we have now. We could have sent an email to the mailing list asking for their feedback(We should do that for this comment thread). We could try to keep the "cfs" repo a place for questions and discussions, or a bunch of other features github has that I am not familiar with. 

But whatever we end up deciding to use it will take a lot of work and commitment to make sure it doesn't fade away.

jphickey, 2021-03-05:
I've come to expect that to some degree - on one hand external users often want to be involved in the process, but they are too busy to participate.  There is nothing stopping a user from subscribing to/watching this repo on github so they can get notified when new discussion topics arise.  But most do not.... maybe concerned that the signal/noise ratio isn't high enough?  In that sense maybe a non-github forum that was devoted to Q&A would improve the S/N ratio and convince external folks to actually look at it.

But it reminds me about that old mantra of bringing a horse to water... we can only do so much.  

I'm not opposed to other forums - whatever discourse/discord forum is created I'll happily go and sign myself in and subscribe to the channel(s) - and maybe it'll gain traction - but old habits die hard.

mogar, 2021-03-08:
For what it's worth, the way NASA cFS teams use github is very unintuitive to me. I tend to think of issues on github as being about low level code changes, so the idea of having discussions like this one on github never crosses my mind (thanks for posting the link to this to the mailing list). If lots of other cFS users are like me, that probably explains why you don't get a lot of people following the repos.

Seems to me the cFS community needs services for:

- internal cFS development discussions
- usage questions
- high level info broadcast from internal devs to users

Right now I'm never sure where to go for any of the above, and end up either posting to the mailing list or trying to see if a question fits as an issue in one of the repos. I've gotten better response rates from the mailing list, so have moved away from making github issues.

I personally would prefer for the services for the three items above to be separate. I notice that I'm less likely to send user-level questions to the mailing list because I perceive it to be mostly for internal dev discussions.

I also think cFS needs a central web page that has way more entry level info in it. I personally get confused about what versions are what, and what works with what, and how to get started. AFAIK, the only place for that info right now is spread out among a bunch of different repos and the mailing list. That makes it super hard to get a new dev up and running with cFS in our org.

If you had a community web page that had entry-level info, you could also put something like @astrogeco's screenshots there. I don't personally have an opinion on slack/discord/whatever. I do think you need to make it easier for people to know whatever decision you pick. I suspect the failure of the slack channel recently is because it was super hard to find if you missed the original emails about it.

I guess my main ask is that there be a single cFS web site where everything else is linked from. Then if I have a question I go there and look around, and eventually find whatever services you decide on.

JackNWhite, 2021-03-09:
The problem with the mailing list is the annoyance of retrieving information from it if you are not actively involved in a discussion. I downloaded each monthly archive and use grep to search them.

What could possibly be the harm in replacing it with a web forum? Everything is kept, everything can be labelled and cross-referenced, it's easy to search, it keeps itself up to date, and everyone has a web browser.

This is a better prospect than Slack, which is an instant messaging service, or a mailing list, which is cumbersome, difficult to search and difficult to cross-reference.

nasabenz, 2021-03-09:
I agree with the problem with the mailing list, it is hard to search.  This leads to similar questions being asked many times.  I think the problem in the past with moving to Stack Overflow is that the mailing list was still there and what was used most.  If we retire the mailing list with an automated reply to move to the new discussion forum then adoption is more likely.

skliper, 2021-06-30:
https://github.com/nasa/cFS/discussions, close?

irowebbn, 2022-12-08:
Just wanted to bump this issue and reiterate what @nasabenz said last year. I think if GitHub discussions is now the preferred forum for discussion, the mailing list should be sunset. I don't mind mailing-lists in general, but the current implementation is very bad. Everyone's corporate email clients constantly break threads, the archive is incomplete, and search is broken. Regardless of what the replacement solution is, I think the fragmentation is causing topics to not get the attention they might get otherwise. GitHub discussions still has some downsides to it (namely, the inability to subscribe to only certain topics/categories), but it would be best to consolidate as much as possible into a single forum.