nasa/cFS #208: Complete Writing High Quality Code for Contributing.MD

ArielSAdamsNASA, 2021-03-04:
**Is your feature request related to a problem? Please describe.**
Provide instructions to users on how to write high quality code. 

**Describe the solution you'd like**
Complete the Writing High Quality Code section to add to the Contributing Guide. 

**Additional context**
The draft so far:

## Writing High-Quality Code

1. Follow cFS code conventions (formatting, symbol naming, file naming, etc) but do *not* change/reformat existing code except to address your changes.

2. For any new API's, add unit tests to cover nominal and off-nominal conditions.

3. Add/edit stubs to the unit test codebase for any new/modified functions.

4. For any changes to existing API's, alter the unit tests to cover the changes (and remove tests made irrelevant due to your changes).

5. Test your code, on a Linux platform (at minimum) --
    _TODO:_ Specific distros/versions/CPU architectures of Linux?

6. Build your code, including unit tests.
    _TODO:_ need "standard" build process.

6. Run the unit tests (all passed, yes?)
    _TODO:_ need "standard" test procedure.

7. Run the main cfs executable (no errors reported, yes?)
    _TODO:_ need "standard" test procedure, including targets.cmake recommendations.

8. ?? Do we expect contributors to run coverage ?? Guessing no.

**Requester Info**
Ariel Adams, ASRC Federal

skliper, 2021-03-04:
One option for 5-9 is to simply provide instructions for how to trigger all the GitHub actions that perform all these activities for the user.  Could also reference the implementations of those actions from the documentation for the definitive/working instructions vs duplicating the instructions in documentation which could get out of sync.  It's even easier if the actions are set to run on all pushes, then a user can simply push a bundle branch to their fork and see that everything passes (or not).

ArielSAdamsNASA, 2021-03-04:
> One option for 5-9 is to simply provide instructions for how to trigger all the GitHub actions that perform all these activities for the user. 

I personally prefer this option.

PR #218 (2021-04-01): Fix #208, Add Writing High Quality Code
**Describe the contribution**
Fix #208 
Add Writing High Quality Code to the Contributing Guide

**Expected behavior changes**
Users should know how to write high quality code and how to use the GitHub Action Workflows. 

**Additional context**
The Writing High Quality Code provides instructions for only the two static code analyzers used in GitHub Actions. 
Should we add instructions for both Deprecated and OMIT_DEPRECATED Build, Test, and Run workflows? They only run when code is pushed to the main branch. If we want to add this, either the workflow ne [... cut]
commit 11e7d6df (2021-04-01): Fix #208, Add Writing High Quality Code
commit b01abf3d (2021-04-01): Merge pull request #218 from ArielSAdamsNASA/Fix-208-high_quality_code

Fix #208, Add Writing High Quality Code
PR #227 (2021-04-02): cFS bundle Integration candidate: 2021-03-30
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

## Describe the contribution
Includes

<https://github.com/nasa/cfe/pull/1258>
<https://github.com/nasa/osal/pull/927>
<https://github.com/nasa/PSP/pull/279>
<https://github.com/nasa/sample_app/pull/145>
<https://github.com/nasa/ci_lab/pull/85>
<https://gith [... cut]
closed 2021-04-02 by astrogeco