nasa/cFS #226: Add links or instructions to pull request squash and ammend section

astrogeco, 2021-03-24:
**Checklist (Please check before submitting)**

* [ ] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ ] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [ ] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
```.md
Once a Pull Request Ready for Review

    Verify the commit message and PR title use the template Fix #XYZ, SHORT_DESCRIPTION.
    Verify there is only one commit message. Squash or amend the commit messages if necessary.
    Verify that the PR passes all checks.
    The project team will label the issue and evaluate the pull request in the weekly configuration control board (CCB) meeting. For more information, visit The cFS CCB Process.
    If the pull request is accepted, it will be merged into cFS.

```

**Describe the solution you'd like**
Add links to instructions of "how" to squash a commit using git rebase or how to ammend a commit

**Describe alternatives you've considered**
Leave as is and address with users on a one-on-one, per PR basis

skliper, 2021-03-25:
Can we do
```
Fix #XYZ, SHORT_DESCRIPTION

LONG_DESCRIPTION (if necessary)
```
and one commit message **per topic**

astrogeco, 2021-03-25:
We might need to expand on the one commit per topic means. Maybe use some of Joe's PRs as examples?

PR #237 (2021-04-22): Fix #226, Add Instructions on How to Squash and Amend Commits
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #226

- Added How to Squash Commits.
- Added How to Amend Commits.
- Added that users may use the format `Fix #XYZ, LONG_DESCRIPTION` for commit messages if necessary. 
- Added to create a commit message p [... cut]
commit 397c6b1d (2021-04-22): Merge pull request #237 from ArielSAdamsNASA/fix-226-add-instructions-to-squash-amend-commits

Fix #226, Add Instructions on How to Squash and Amend Commits
commit d15f7bff (2021-04-22): Fix #226, Add Instructions on How to Squash and Amend Commits
closed 2021-04-22 by astrogeco