nasa/cFS #229: SB Message data

carte731, 2021-03-26:
hello everyone,

Sorry for the basic question. i’m new to cFS - I’ve tried for a few days now and having a some difficulty with SB messages. 

I’ve read through the development guide and API - and I can’t figure out how to properly attach data to a SB message. 

I can send and receive messages, but when I mount the pointer to a variable - all the data assigned in my structs are empty. 

How do I properly mount custom data to my SB messages?

Corey Carter - NASA Pathways

dmccomas, 2021-03-26:
I can’t remember if the opensatkit ‘hello world code inspection’ video has
the level of detail you need. Feel free to email me at ***@***.***
and I’d be happy to meet.

Dave

carte731, 2021-03-30:
I figured out the problem, it was a combination of a typecasting error on the receiving end and an error with my app table. The table error was interfering with the start up sequence and it was effecting message delivery in my cFS build. I addressed the issues and the struct contents and their pointers are now being generated and transmitted properly. Thank you for reaching out dmccomas - I appreciate the help and I'll check out the OpenSatKit video tomorrow!

closed 2021-03-30 by carte731