nasa/cFS #244: Guide to Adding New Toolchains

JackNWhite, 2021-04-19:
Hi,

Is there a guide anywhere for adding new toolchains? I want to add an
Aarch64 Linux toolchain, but I am confused by the example toolchains.

For example:
- What do these mean?
    - CMAKE_FIND_ROOT_PATH_MODE_PROGRAM
    - CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    - CMAKE_FIND_ROOT_PATH_MODE_INCLUDE
    - CMAKE_PREFIX_PATH
- What needs to be added, and what not? The arm-cortexa8_neon-linux-gnueabi example only specifies the compilers, whereas the i686-rtems example specifies `ld`, `as`, `strip`, `nm`, `ar`, `objdump` and `objcpy` too.

Does anyone already have a toolchain script for Linux on Aarch64?

Many thanks,

Jack

JackNWhite, 2021-04-19:
I have set up the following `toolchain-aarch64-linux-gnu.cmake`:
```
# Toolchain file for the Xilinx ZCU102 Cortex-A53 processors

# Basic cross system configuration
SET(CMAKE_SYSTEM_NAME           Linux)
SET(CMAKE_SYSTEM_VERSION        1)
SET(CMAKE_SYSTEM_PROCESSOR      aarch64)

# adjust these settings to where the cross compiler actually resides
SET(CMAKE_C_COMPILER            "/usr/bin/aarch64-linux-gnu-gcc")
SET(CMAKE_CXX_COMPILER          "/usr/bin/aarch64-linux-gnu-cpp")

# search for programs in the build host directories
SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM   NEVER)

# for libraries and headers in the target directories
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY   ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE   ONLY)

# these settings are specific to cFE/OSAL and determines which 
# abstraction layers are built when using this toolchain
SET(CFE_SYSTEM_PSPNAME      "pc-linux")
SET(OSAL_SYSTEM_BSPTYPE     "pc-linux")
SET(OSAL_SYSTEM_OSTYPE      "posix")
```
I have added `SET(cpu1_SYSTEM aarch64-linux-gnu)` to `targets.cmake`.

Nonetheless, when I run straight up `VERBOSE=1 make`, I can see that the compiler used is `/usr/bin/cc`.

jphickey, 2021-04-19:
Hi Jack, here are a few suggestions:
- Documentation about the various toolchain variables are here: https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html
- The `CMAKE_C_COMPILER` setting is absolutely required since the cross-compiled code is almost entirely C.  For the others you may be able to get away without them but I recommend specifying all of them (e.g. OBJDUMP, etc) unless you have a reason not to.  In my experience with ELF the supplemental tools are often interchangeable because its a generic binary format.  But still you might run into compatibility issues, so better to specify all of them, I think.
- Toolchain info is cached by CMake.  Meaning that after the initial CMake run, the values are retained, and are NOT updated if you change the file later to specify a different toolchain.  To make it use the new values the best way is to completely remove your build tree and regenerate from a clean slate, or you can create a new/separate build tree for the other target.

JackNWhite, 2021-04-19:
I have got my toolchain working.

closed 2021-04-19 by JackNWhite