nasa/cFS #245: Update all community email management references for new servcie

skliper, 2021-04-20:
**Checklist (Please check before submitting)**

* [ x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ x] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [ x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
Community email system is transitioning (currently May 8th target), which changes join/leave process.

**Describe the solution you'd like**
Update instructions in sync with system change over:
```
To subscribe to [INSERT LIST NAME]: 
Send an e-mail to LISTNAME-join@lists.nasa.gov (no subject or text in the body is required).

To unsubscribe to [INSERT LIST NAME]:
Send an e-mail to LISTNAME-leave@lists.nasa.gov (no subject or text in the body is required).

Having trouble subscribing or unsubscribing from a list?
Get help by sending an e-mail to the list owner at LISTNAME-owner@lists.nasa.gov
```

**Describe alternatives you've considered**
#207

**Additional context**
Note a web interface is pending.  Not clear if it will support review of archives by members.

**Requester Info**
Jacob Hageman

astrogeco, 2022-05-25:
I think https://github.com/nasa/cFS/pull/258 fixed this. What are we missing?

skliper, 2022-05-25:
Just tested the methods described in #258 and it works for me.  Linked and closing.

closed 2022-05-25 by skliper