nasa/cFS #246: Relationship of cFS CPU number to Linux CPU number

JackNWhite, 2021-04-21:
Is there a definite relationship between the cFS CPU number and the number given by the Linux `ps` command?

I am working from the files in `cfe/sample_defs`, where the demo purports to run on CPU 1. It seems to me that this could mean either CPU 0 or CPU 1, in the Linux labelling system, but it is not clear which. When I actually run `ps`, however, my cFS instance is running on CPU 2, i.e. the third CPU.

Ta,
Jack

skliper, 2021-04-21:
No, cpuname within the cFS build system is just a name used to associate settings and configure build directories:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/cmake/sample_defs/targets.cmake#L8-L9
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/cmake/sample_defs/targets.cmake#L100-L101

The name can really be any prefix and has no influence on processor (cpu) affinity.  Currently the cFS framework does not set processor (cpu) affinity, so it's up to the OS to manage.  For Linux see taskset.

There is internal work in progress to add support for symmetric multicore processing (SMP) in relation to managing processor affinity from within the framework but that is not in the development branch yet.

Thanks,
Jake

JackNWhite, 2021-04-26:
Thanks, @skliper

closed 2021-04-26 by JackNWhite