nasa/cFS #247: Separate app and libraries

astrogeco, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
Having libraries in the apps folder can be confusing especially for beginners.

**Describe the solution you'd like**
New directory called "lib" or "libs" where libraries can "live"

**Describe alternatives you've considered**
Add documentation justifying why it "makes sense" to keep libraries in the apps directory

jphickey, 2021-04-22:
This already exists, there is a "libs" dir (equivalent to apps) as part of the search path already.  We just don't use it in the bundle.  All we need to do is move e.g. "sample_lib" submodule from `apps` into `libs` in the bundle and it should just work.

astrogeco, 2021-04-23:
Awesome! I can try that next week

PR #335 (2021-08-17): Fix #247, Move sample_libs into libraries dir
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Creates a new directory "libs" at the cFS bundle level to put libraries in. This will create a little more clarity between applications and libraries

Fixes #247

**Testing performed**
CI Ubuntu

**Expected behavior changes**
N [... cut]
commit 75557733 (2021-08-17): Fix #247, Move sample_libs into libraries dir

Set default branch for submodules in .gitmodules to "main"
commit e631fbb8 (2021-08-17): Merge pull request #335 from astrogeco/247-use-libs-dir

Fix #247, Move sample_libs into libraries dir
closed 2021-08-17 by astrogeco