nasa/cFS #266: RPI4 compile error

mkressel, 2021-06-12:
Hi, I'm trying to compile the cFS on a Raspberry PI 4. I've followed the instructions [here](http://alanstechnotes.blogspot.com/2018/11/how-to-compile-core-flight-system-on.html?m=1), to modify the system processor from i686 to arm for cmake. (The other edits seemed deprecated). 

When compiling, I get the following error:

```Scanning dependencies of target cfe_ts_crc
[ 85%] Building C object tools/tblCRCTool/CMakeFiles/cfe_ts_crc.dir/cfe_ts_crc.c.o
/home/pi/src/cFS/tools/tblCRCTool/cfe_ts_crc.c: In function ‘main’:
/home/pi/src/cFS/tools/tblCRCTool/cfe_ts_crc.c:156:77: error: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘ssize_t’ {aka ‘int’} [-Werror=format=]
     printf("\nTable File Name:            %s\nTable Size:                 %ld Bytes\nExpected TS Validation CRC: "
                                                                           ~~^
                                                                           %d
/home/pi/src/cFS/tools/tblCRCTool/cfe_ts_crc.c:158:21:
            argv[1], fileSize, fileCRC);
                     ~~~~~~~~
cc1: all warnings being treated as errors
make[7]: *** [tools/tblCRCTool/CMakeFiles/cfe_ts_crc.dir/build.make:63: tools/tblCRCTool/CMakeFiles/cfe_ts_crc.dir/cfe_ts_crc.c.o] Error 1
make[6]: *** [CMakeFiles/Makefile2:622: tools/tblCRCTool/CMakeFiles/cfe_ts_crc.dir/all] Error 2
make[5]: *** [Makefile:141: all] Error 2
make[4]: *** [CMakeFiles/mission-install.dir/build.make:57: CMakeFiles/mission-install] Error 2
make[3]: *** [CMakeFiles/Makefile2:297: CMakeFiles/mission-install.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:304: CMakeFiles/mission-install.dir/rule] Error 2
make[1]: *** [Makefile:227: mission-install] Error 2
make: *** [Makefile:122: install] Error 2
```

Are there other edit(s) I need to make to a cmake file to get cFS to compile on the RPI4? Thanks!

mkressel, 2021-06-12:
FYI, I was able to get it to compile on the RPI4 by changing the "%ld" to "%d" on line 156 of tools/tblCRCTool/cfe_ts_crc.c. The cFS now compiles and runs successfully.

jphickey, 2021-06-14:
This is an issue in the tblCRCTool, it needs a cast here.  Recommend to transfer this issue to that repo so a patch can be done.