nasa/cFS #267: Add a bot that automatically posts the current repository hash when issues are created

astrogeco, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
When looking at really old tickets it is hard to know what they refer to 

**Describe the solution you'd like**
Bot that automatically gets the current git-sha when an issue is posted. 

**Describe alternatives you've considered**
Have a section that asks which version of the component the user is reporting the bug against

jeffstjean, 2021-06-24:
In my opinion, it would be nice to have the user explicitly reference their version (for cFE, OSAL, and the app of interest). I know many (most?) cFS bundles that are nowhere near the current release and it might just be "another check in the box" when someone is posting a question instead of forcing them to be aware of versioning mismatch (an easy mistake to make!).

astrogeco, 2021-06-25:
> In my opinion, it would be nice to have the user explicitly reference their version (for cFE, OSAL, and the app of interest). I know many (most?) cFS bundles that are nowhere near the current [...]

thanks for the input! what do you mean by "another check in the box"? One challenge as I think more about this is that the bot would only report the repo's sha as opposed to the user's version. Maybe an easier way to do this is to have a command kind of like `cfs --version` that makes this as easy as possible for everyone. We already have this on startup but it might be nice...

jeffstjean, 2021-06-25:
Sorry I think I slightly misunderstood the bot. I thought it would provide the poster with the SHA for them to paste into the issue itself. I do like the idea of having a CLI that can provide a host of tools, including a version output. I had spoken to a couple of the community members of having some central repository (Github!) that a CLI could pull the right version of community apps down from to make installation a breeze. The Node Package Manager (NPM) could be a good template to jump off from!