nasa/cFS #268: Add CI workflow/configuration that enables strict type checking

jphickey, 2021-06-16:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
The cFS CI scripts currently build with lax/weak type checking, which is the distribution default (in sample_defs) as it is more backward compatible.

**Describe the solution you'd like**
At least one CI build should enable this in `global_build_options.cmake` to enable strict type checking:

    set(MISSION_RESOURCEID_MODE STRICT)

**Additional context**
It's easy to add code that will fail w/strict type checks, unless this is enabled in CI to catch it.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

astrogeco, 2021-06-23:
Can this be done on the command line or does it require a file edit (via sed or another tool)?

skliper, 2022-05-03:
Strict type-checking in OSAL is enabled with `OMIT_DEPRECATED=true` which we do in multiple workflows including build-cfs,yml build-run-app.yml unit-test-coverage.yml, etc.  Closing as duplicate of that work.

closed 2022-05-03 by skliper