nasa/cFS #275: Error installing cFS bootes

kaushikvarma28, 2021-06-23:
**Checklist (Please check before submitting)**

* [ ] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ ] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
Hi, I am trying to install cFS bootes and I am getting an error while installing it. I tried installing the cFS normal version and it installed successfully but, when I try bootes it is giving the error as shown below.

**To Reproduce**
Steps I followed:
git clone --recursive https://github.com/astrogeco/cFS.git 
cd cFS
git checkout checksum-bootes-working
export SIMULATION=native
make prep
make install (The error shows up here)

**Code snips**
pi@raspberrypi:~/cFS $ make install
make --no-print-directory -C "build" DESTDIR="/home/pi/cFS/build" mission-install
[ 50%] Building C object tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o
In file included from /home/pi/cFS/cfe/modules/core_api/fsw/inc/cfe_tbl_extern_typedefs.h:41,
                 from /home/pi/cFS/cfe/modules/core_api/fsw/inc/cfe_tbl_filedef.h:51,
                 from /home/pi/cFS/tools/elf2cfetbl/elf2cfetbl.c:42:
/home/pi/cFS/cfe/modules/core_api/fsw/inc/cfe_es_extern_typedefs.h:532:29: error: ‘CFE_MISSION_ES_CDS_MAX_FULL_NAME_LEN’ undeclared here (not in a function); did you mean ‘CFE_MISSION_ES_CDS_MAX_NAME_LEN’?
     char               Name[CFE_MISSION_ES_CDS_MAX_FULL_NAME_LEN]; /**< \brief Processor Unique Name of CDS */
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                             CFE_MISSION_ES_CDS_MAX_NAME_LEN
/home/pi/cFS/cfe/modules/core_api/fsw/inc/cfe_es_extern_typedefs.h:567:36: error: ‘CFE_MISSION_ES_POOL_MAX_BUCKETS’ undeclared here (not in a function); did you mean ‘CFE_MISSION_ES_PERF_MAX_IDS’?
     CFE_ES_BlockStats_t BlockStats[CFE_MISSION_ES_POOL_MAX_BUCKETS]; /**< \cfetlmmnemonic \ES_BLKSTATS
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                    CFE_MISSION_ES_PERF_MAX_IDS
make[4]: *** [tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/build.make:63: tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o] Error 1
make[3]: *** [CMakeFiles/Makefile2:638: tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:209: CMakeFiles/mission-install.dir/rule] Error 2
make[1]: *** [Makefile:227: mission-install] Error 2
make: *** [Makefile:123: install] Error 2

**System observed on:**
 - Hardware: Raspberry pi 4
 - OS: Raspbian

**Reporter Info**
Name: Kaushik Varma Rudraraju
College: Colorado State University

skliper, 2021-06-23:
Looks like you are cloning from astrogeco's fork.  You may have better luck using the bootes.x branch from the nasa repo and follow the quick-start instructions within that branch's README.md.

jphickey, 2021-06-23:
That error is a mismatch between the "mission_cfg.h" and/or "platform_cfg.h" file and the version of CFE that you are attempting to compile.

- you should use the bootes.x branch from the official repo as noted in previous comment
- check the "cfe/cmake/sample_defs" directory and at least compare to the one you are currently using.  If you are upgrading from an older release e.g. acquila, then note that some config items are new in bootes and others were removed.  You'll have to update your config to reflect this.  If you attempt to build bootes directly with a config from an older version, you'll likely get compile errors like the one shown.

commit 9b842c17 (2021-03-31): IC:2021-03-30, Add remaining PRs and FastTrakcs

nasa/cfe#1260 - Fix #1259, adds --branch-coverage option to genhtml line in lcov Makefile rule

nasa/osal#915 - Fix #888, better return codes from OS_SymbolTableDump_Impl
nasa/osal#926 - Fix #853, remove OS_TaskRegister
nasa/osal#935 - Fix #934, Remove unused SCRIPT_MODE flag
nasa/osal#930 - Fix #929, use test function for osal_id_t

nasa/psp#276 - Fix #275, Adds a UT stub for CFE_PSP_GetKernelTextSegmentInfo

nasa/ci_lab#84 - Fix #73, use zero copy API
nasa/sample_app#143 - Fix #142, scrub header guards